kakopappa/arduino-esp8266-alexa-wemo-switch #1: no working

ekerp, 2016-08-03:
I upload it to wemos 8266 and echo dows no activate relay

kakopappa, 2016-08-15:
Updated the missing relay code. Thanks for creating an issue

closed 2016-08-15 by kakopappa

ekerp, 2016-08-15:
Thanks now it works
Question  I need to learn  to code arduino any good place to start.?

kakopappa, 2016-08-16:
> <javascript:_e(%7B%7D,'cvml','ekerp@hotmail.com');>>, Author <
> kakopappa/arduino-esp8266-alexa-wemo-switch/issues/1#

This is my first real Arduino project as well. Prior to this I had very
little experience in Arduino. If you want to get into Arduino programming I
recommend you to start with basics like turning on of LEDs and then move to
more advanced stuff. I used YouTube alot to search for this and that. You
can find almost anything there

Louis Rossman has good tutorials on YouTube  to explain components like
Transistors, capacitors in plain English

https://youtu.be/4prl_6eSJLY

Good luck, let me know if you need any help

ekerp, 2016-08-16:
thanks for your big help ....another thing how can i add a power bottom to this project so i can turn it from local switch too.

[Image by FlamingText.com]<http://www.flamingtext.com/hmail.html>

https://youtu.be/4prl_6eSJLY

kakopappa, 2016-08-18:
Look for a push button example
https://www.arduino.cc/en/Tutorial/Button