kakopappa/arduino-esp8266-alexa-wemo-switch #2: Openhab

shenery, 2016-09-20:
Hi, great project! I was wondering how easy this would be to implement into openhab?
Also what are the wifi settings that I have to change and what firmware flasher would did you use?
I have found http://www.xess.com/blog/esp8266-reflash/
Would that work ok?

Kind regards
Steven

shenery, 2016-09-24:
Ok. I've changed the ssid and wifi password but Alexa can't find it. Should there be a blue light on the wemos? And should the relay power the wemos board from the mains?

shenery, 2016-09-24:
Also, I have soldered all the pins from the wemos to the relay board instead of just the ones you mentioned. Not sure if that's my problem?

kakopappa, 2016-09-25:
I am not sure whether you have the right setup. So

First, Make sure you have the correct parts. I did this project using

WeMos D1-mini ($4.00) http://www.aliexpress.com/store/product/D1-mini-Mini-NodeMcu-4M-bytes-Lua-WIFI-Internet-of-Things-development-board-based-ESP8266/1331105_32529101036.html

WeMos Relay ($1.90) http://www.aliexpress.com/store/product/Relay-Shield-for-WeMos-D1-mini-button/1331105_32596395175.html

Once you connect both devices it should look like this
https://1drv.ms/i/s!AlN6Rnygw-lfgrR5nLNXfdMJslvYoA

You said you have soldered them, make sure they are soldered correctly.

2. then change the Wifi settings in sinric.ino and flash it

3. When the app starts, it will start listening for UDP broadcasts on port 1900. If you open the Arduino debug serial monitor you should see messages like this. 

########## Responding to  /upnp/control/basicevent1

Even if you are using another device, when you scan for devices it should print a log like this. 

Can you see this ?

shenery, 2016-09-25:
I have the same device as you and it looks identical to your setup. I will check arduino debug tonight. Can this be used with the belkin android app or only with Amazon echo? I have the the live wire connected to the normally open but I dont see a light on the wemos. Should the relay power the wemos board?

shenery, 2016-09-25:
I think I set up serial debug library and I didn't get any errors when I compiled it. heres what I got.

C:\Users\Shenery\Documents\Arduino\arduino-esp8266-alexa-wemo-switch-master\sinric\sinric.ino:105:28: warning: character constant too long for its type [enabled by default]

         if(request.indexOf('M-SEARCH') > 0) {

                            ^

C:\Users\Shenery\Documents\Arduino\arduino-esp8266-alexa-wemo-switch-master\sinric\sinric.ino: In function 'void loop()':

C:\Users\Shenery\Documents\Arduino\arduino-esp8266-alexa-wemo-switch-master\sinric\sinric.ino:105:38: warning: large integer implicitly truncated to unsigned type [-Woverflow]

         if(request.indexOf('M-SEARCH') > 0) {

                                      ^

Sketch uses 279,885 bytes (26%) of program storage space. Maximum is 1,044,464 bytes.
Global variables use 48,500 bytes (59%) of dynamic memory, leaving 33,420 bytes for local variables. Maximum is 81,920 bytes.

the only thing I've changed is the ssid and WiFi password. do I have to change anything and exactly  else in the code? also my soldering looks good and exactly the same as your picture.

thanks for your help and patience I am very new to wemos and Arduino ide.

shenery, 2016-09-27:
Do you think it could be that I haven't added the library's?
include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <WiFiUdp.h>
#include <functional>

Which library's did you have to add?

kakopappa, 2016-09-27:
Aha. you need to add esp8266 library to Ardino IDE. If you haven't done that check here

https://forum.wemos.cc/topic/24/where-do-i-get-esp8266wifi-h/3

shenery, 2016-09-27:
I have installed https://github.com/esp8266/Arduino/tree/master/tools
Executed python get.py so should I mot already have the libraries? When I
open serial monitor it shows nothing just a send button?
I'm really not sure what do try.

Steven

shenery, 2016-09-28:
I followed this guide https://www.wemos.cc/tutorial/get-started-arduino.html to the letter and It seems to compile fine but I can't see the wemos board on my network at all after it. And also when I open serial monitor and connect to pc I don't see anything at all. Just 2 white boxes with no text and a send button.

kakopappa, 2016-09-28:
Before flashing sinric.io i recommend you to try this first

1. Write a hello world program and print it on serial monitor.

void setup()
{
  Serial.begin(9600);  
  Serial.println("Setup..");
}

void loop() 
{
  Serial.println("Hello World..");
  delay(1000); // waits for a second
}

Open the serial monitor. Change the baud rate to 9600 in the drop down below

Do you see anything ?

shenery, 2016-09-28:
serial monitor didn't show anything but when I tried to verify/compile I got this message

Arduino: 1.6.12 (Windows 10), Board: "WeMos D1 R2 & mini, 80 MHz, 921600, 4M (3M SPIFFS)"

C:\Program Files\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\Arduino\hardware -tools C:\Program Files\Arduino\tools-builder -tools C:\Program Files\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files\Arduino\libraries -libraries C:\Users\Shenery\Documents\Arduino\libraries -fqbn=esp8266com:esp8266:d1_mini:CpuFrequency=80,UploadSpeed=921600,FlashSize=4M3M -ide-version=10612 -build-path C:\Users\Shenery\AppData\Local\Temp\arduino_build_100668 -warnings=none -prefs=build.warn_data_percentage=75 -verbose C:\Users\Shenery\AppData\Local\Temp\untitled294378707.tmp\sketch_sep28b\sketch_sep28b.ino
C:\Program Files\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files\Arduino\hardware -tools C:\Program Files\Arduino\tools-builder -tools C:\Program Files\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files\Arduino\libraries -libraries C:\Users\Shenery\Documents\Arduino\libraries -fqbn=esp8266com:esp8266:d1_mini:CpuFrequency=80,UploadSpeed=921600,FlashSize=4M3M -ide-version=10612 -build-path C:\Users\Shenery\AppData\Local\Temp\arduino_build_100668 -warnings=none -prefs=build.warn_data_percentage=75 -verbose C:\Users\Shenery\AppData\Local\Temp\untitled294378707.tmp\sketch_sep28b\sketch_sep28b.ino
Using board 'd1_mini' from platform in folder: C:\Program Files\Arduino\hardware\esp8266com\esp8266
Using core 'esp8266' from platform in folder: C:\Program Files\Arduino\hardware\esp8266com\esp8266
Detecting libraries used...
"C:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/sdk/include" "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/sdk/lwip/include" "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\Shenery\AppData\Local\Temp\arduino_build_100668/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10612 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_WEMOS_D1MINI"  -DESP8266 "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266\cores\esp8266" "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266\variants\d1_mini" "C:\Users\Shenery\AppData\Local\Temp\arduino_build_100668\sketch\sketch_sep28b.ino.cpp" -o "nul"
Generating function prototypes...
"C:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/sdk/include" "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/sdk/lwip/include" "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\Shenery\AppData\Local\Temp\arduino_build_100668/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10612 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_WEMOS_D1MINI"  -DESP8266 "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266\cores\esp8266" "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266\variants\d1_mini" "C:\Users\Shenery\AppData\Local\Temp\arduino_build_100668\sketch\sketch_sep28b.ino.cpp" -o "C:\Users\Shenery\AppData\Local\Temp\arduino_build_100668\preproc\ctags_target_for_gcc_minus_e.cpp"
"C:\Program Files\Arduino\tools-builder\ctags\5.8-arduino10/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\Users\Shenery\AppData\Local\Temp\arduino_build_100668\preproc\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/sdk/include" "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/sdk/lwip/include" "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\Shenery\AppData\Local\Temp\arduino_build_100668/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10612 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_WEMOS_D1MINI"  -DESP8266 "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266\cores\esp8266" "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266\variants\d1_mini" "C:\Users\Shenery\AppData\Local\Temp\arduino_build_100668\sketch\sketch_sep28b.ino.cpp" -o "C:\Users\Shenery\AppData\Local\Temp\arduino_build_100668\sketch\sketch_sep28b.ino.cpp.o"
C:\Users\Shenery\AppData\Local\Temp\untitled294378707.tmp\sketch_sep28b\sketch_sep28b.ino: In function 'void loop()':

sketch_sep28b:11: error: expected '}' at end of input

  delay(1000); // waits for a second

             ^

exit status 1
expected '}' at end of input

kakopappa, 2016-09-29:
it seems you have missed } at the end of the file?

if it does not work  try 
https://www.arduino.cc/en/Serial/Print

shenery, 2016-09-29:
ok I added that and got this.

Arduino: 1.6.12 (Windows 10), Board: "WeMos D1 R2 & mini, 80 MHz, 921600, 4M (3M SPIFFS)"

C:\Program Files\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\Arduino\hardware -tools C:\Program Files\Arduino\tools-builder -tools C:\Program Files\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files\Arduino\libraries -libraries C:\Users\Shenery\Documents\Arduino\libraries -fqbn=esp8266com:esp8266:d1_mini:CpuFrequency=80,UploadSpeed=921600,FlashSize=4M3M -ide-version=10612 -build-path C:\Users\Shenery\AppData\Local\Temp\arduino_build_191528 -warnings=all -prefs=build.warn_data_percentage=75 -verbose C:\Users\Shenery\Documents\Arduino\sketch_sejkk\sketch_sejkk.ino
C:\Program Files\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files\Arduino\hardware -tools C:\Program Files\Arduino\tools-builder -tools C:\Program Files\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files\Arduino\libraries -libraries C:\Users\Shenery\Documents\Arduino\libraries -fqbn=esp8266com:esp8266:d1_mini:CpuFrequency=80,UploadSpeed=921600,FlashSize=4M3M -ide-version=10612 -build-path C:\Users\Shenery\AppData\Local\Temp\arduino_build_191528 -warnings=all -prefs=build.warn_data_percentage=75 -verbose C:\Users\Shenery\Documents\Arduino\sketch_sejkk\sketch_sejkk.ino
Using board 'd1_mini' from platform in folder: C:\Program Files\Arduino\hardware\esp8266com\esp8266
Using core 'esp8266' from platform in folder: C:\Program Files\Arduino\hardware\esp8266com\esp8266
Detecting libraries used...
"C:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/sdk/include" "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/sdk/lwip/include" "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\Shenery\AppData\Local\Temp\arduino_build_191528/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10612 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_WEMOS_D1MINI"  -DESP8266 "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266\cores\esp8266" "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266\variants\d1_mini" "C:\Users\Shenery\AppData\Local\Temp\arduino_build_191528\sketch\sketch_sejkk.ino.cpp" -o "nul"
Generating function prototypes...
"C:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/sdk/include" "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/sdk/lwip/include" "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\Shenery\AppData\Local\Temp\arduino_build_191528/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10612 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_WEMOS_D1MINI"  -DESP8266 "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266\cores\esp8266" "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266\variants\d1_mini" "C:\Users\Shenery\AppData\Local\Temp\arduino_build_191528\sketch\sketch_sejkk.ino.cpp" -o "C:\Users\Shenery\AppData\Local\Temp\arduino_build_191528\preproc\ctags_target_for_gcc_minus_e.cpp"
"C:\Program Files\Arduino\tools-builder\ctags\5.8-arduino10/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\Users\Shenery\AppData\Local\Temp\arduino_build_191528\preproc\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/sdk/include" "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/sdk/lwip/include" "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\Shenery\AppData\Local\Temp\arduino_build_191528/core" -c -Wall -Wextra -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10612 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_WEMOS_D1MINI"  -DESP8266 "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266\cores\esp8266" "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266\variants\d1_mini" "C:\Users\Shenery\AppData\Local\Temp\arduino_build_191528\sketch\sketch_sejkk.ino.cpp" -o "C:\Users\Shenery\AppData\Local\Temp\arduino_build_191528\sketch\sketch_sejkk.ino.cpp.o"
C:\Users\Shenery\Documents\Arduino\sketch_sejkk\sketch_sejkk.ino: In function 'void setup()':

sketch_sejkk:13: error: redefinition of 'void setup()'

 void setup() {

      ^

sketch_sejkk:1: error: 'void setup()' previously defined here

 void setup()

      ^

C:\Users\Shenery\Documents\Arduino\sketch_sejkk\sketch_sejkk.ino: In function 'void loop()':

sketch_sejkk:18: error: redefinition of 'void loop()'

 void loop() {

      ^

sketch_sejkk:8: error: 'void loop()' previously defined here

 void loop() 

      ^

exit status 1
redefinition of 'void setup()'

shenery, 2016-09-29:
I also tried serial print and got this.

Arduino: 1.6.12 (Windows 10), Board: "WeMos D1 R2 & mini, 80 MHz, 921600, 4M (3M SPIFFS)"

C:\Program Files\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\Arduino\hardware -tools C:\Program Files\Arduino\tools-builder -tools C:\Program Files\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files\Arduino\libraries -libraries C:\Users\Shenery\Documents\Arduino\libraries -fqbn=esp8266com:esp8266:d1_mini:CpuFrequency=80,UploadSpeed=921600,FlashSize=4M3M -ide-version=10612 -build-path C:\Users\Shenery\AppData\Local\Temp\arduino_build_645375 -warnings=all -prefs=build.warn_data_percentage=75 -verbose C:\Users\Shenery\AppData\Local\Temp\arduino_modified_sketch_485513\huikk.ino
C:\Program Files\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files\Arduino\hardware -tools C:\Program Files\Arduino\tools-builder -tools C:\Program Files\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files\Arduino\libraries -libraries C:\Users\Shenery\Documents\Arduino\libraries -fqbn=esp8266com:esp8266:d1_mini:CpuFrequency=80,UploadSpeed=921600,FlashSize=4M3M -ide-version=10612 -build-path C:\Users\Shenery\AppData\Local\Temp\arduino_build_645375 -warnings=all -prefs=build.warn_data_percentage=75 -verbose C:\Users\Shenery\AppData\Local\Temp\arduino_modified_sketch_485513\huikk.ino
Using board 'd1_mini' from platform in folder: C:\Program Files\Arduino\hardware\esp8266com\esp8266
Using core 'esp8266' from platform in folder: C:\Program Files\Arduino\hardware\esp8266com\esp8266
Detecting libraries used...
"C:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/sdk/include" "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/sdk/lwip/include" "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\Shenery\AppData\Local\Temp\arduino_build_645375/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10612 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_WEMOS_D1MINI"  -DESP8266 "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266\cores\esp8266" "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266\variants\d1_mini" "C:\Users\Shenery\AppData\Local\Temp\arduino_build_645375\sketch\huikk.ino.cpp" -o "nul"
Generating function prototypes...
"C:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/sdk/include" "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/sdk/lwip/include" "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\Shenery\AppData\Local\Temp\arduino_build_645375/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10612 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_WEMOS_D1MINI"  -DESP8266 "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266\cores\esp8266" "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266\variants\d1_mini" "C:\Users\Shenery\AppData\Local\Temp\arduino_build_645375\sketch\huikk.ino.cpp" -o "C:\Users\Shenery\AppData\Local\Temp\arduino_build_645375\preproc\ctags_target_for_gcc_minus_e.cpp"
"C:\Program Files\Arduino\tools-builder\ctags\5.8-arduino10/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\Users\Shenery\AppData\Local\Temp\arduino_build_645375\preproc\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/sdk/include" "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/sdk/lwip/include" "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\Shenery\AppData\Local\Temp\arduino_build_645375/core" -c -Wall -Wextra -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10612 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_WEMOS_D1MINI"  -DESP8266 "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266\cores\esp8266" "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266\variants\d1_mini" "C:\Users\Shenery\AppData\Local\Temp\arduino_build_645375\sketch\huikk.ino.cpp" -o "C:\Users\Shenery\AppData\Local\Temp\arduino_build_645375\sketch\huikk.ino.cpp.o"
C:\Users\Shenery\AppData\Local\Temp\arduino_modified_sketch_485513\huikk.ino: In function 'void setup()':

huikk:47: error: redefinition of 'void setup()'

 }void setup() {

       ^

huikk:6: error: 'void setup()' previously defined here

 void setup() {

      ^

C:\Users\Shenery\AppData\Local\Temp\arduino_modified_sketch_485513\huikk.ino: In function 'void loop()':

huikk:52: error: redefinition of 'void loop()'

 void loop() {

      ^

huikk:10: error: 'void loop()' previously defined here

 void loop() {  

      ^

exit status 1
redefinition of 'void setup()'

shenery, 2016-09-29:
ok i know im writing a lot but i added the url to board manager and can now see the wemos on my network. i get a lot of errors though.

Arduino: 1.6.12 (Windows 10), Board: "WeMos D1 R2 & mini, 80 MHz, 921600, 4M (3M SPIFFS)"

C:\Program Files\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\Arduino\hardware -tools C:\Program Files\Arduino\tools-builder -tools C:\Program Files\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files\Arduino\libraries -libraries C:\Users\Shenery\Documents\Arduino\libraries -fqbn=esp8266com:esp8266:d1_mini:CpuFrequency=80,UploadSpeed=921600,FlashSize=4M3M -ide-version=10612 -build-path C:\Users\Shenery\AppData\Local\Temp\arduino_build_161554 -warnings=all -prefs=build.warn_data_percentage=75 -verbose C:\Users\Shenery\Documents\Arduino\sinric\sinric.ino
C:\Program Files\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files\Arduino\hardware -tools C:\Program Files\Arduino\tools-builder -tools C:\Program Files\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files\Arduino\libraries -libraries C:\Users\Shenery\Documents\Arduino\libraries -fqbn=esp8266com:esp8266:d1_mini:CpuFrequency=80,UploadSpeed=921600,FlashSize=4M3M -ide-version=10612 -build-path C:\Users\Shenery\AppData\Local\Temp\arduino_build_161554 -warnings=all -prefs=build.warn_data_percentage=75 -verbose C:\Users\Shenery\Documents\Arduino\sinric\sinric.ino
Using board 'd1_mini' from platform in folder: C:\Program Files\Arduino\hardware\esp8266com\esp8266
Using core 'esp8266' from platform in folder: C:\Program Files\Arduino\hardware\esp8266com\esp8266
Detecting libraries used...
"C:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/sdk/include" "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/sdk/lwip/include" "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\Shenery\AppData\Local\Temp\arduino_build_161554/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10612 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_WEMOS_D1MINI"  -DESP8266 "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266\cores\esp8266" "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266\variants\d1_mini" "C:\Users\Shenery\AppData\Local\Temp\arduino_build_161554\sketch\sinric.ino.cpp" -o "nul"
"C:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/sdk/include" "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/sdk/lwip/include" "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\Shenery\AppData\Local\Temp\arduino_build_161554/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10612 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_WEMOS_D1MINI"  -DESP8266 "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266\cores\esp8266" "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266\variants\d1_mini" "-IC:\Users\Shenery\Documents\Arduino\libraries\ESP8266WiFi\src" "C:\Users\Shenery\AppData\Local\Temp\arduino_build_161554\sketch\sinric.ino.cpp" -o "nul"
"C:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/sdk/include" "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/sdk/lwip/include" "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\Shenery\AppData\Local\Temp\arduino_build_161554/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10612 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_WEMOS_D1MINI"  -DESP8266 "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266\cores\esp8266" "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266\variants\d1_mini" "-IC:\Users\Shenery\Documents\Arduino\libraries\ESP8266WiFi\src" "-IC:\Users\Shenery\Documents\Arduino\libraries\ESP8266WebServer\src" "C:\Users\Shenery\AppData\Local\Temp\arduino_build_161554\sketch\sinric.ino.cpp" -o "nul"
"C:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/sdk/include" "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/sdk/lwip/include" "-IC:\Program Files\Arduino\hardware\esp8266com\esp8266/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\Shenery\AppData\Local\Temp\arduino_build_161554/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fd [... cut ...]
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed