kakopappa/arduino-esp8266-alexa-wemo-switch #3: Multiple devices?

Lobojpm, 2016-10-03:
This is really cool how it emulates the Belkin Wemo Switches.  That is a great step, and can avoid having to use intermediate proxies like IFTTT to get from the Echo to the ESP8266!

My question is what it we want to have multiple devices running this code on the network?  

I think I saw the spot to change or set the device name in the code (is it called "box"?).  If I wanted to have multiple devices, would I give them new names here?  Would I need to change other items in the code for some kind of device ID?  

Thanks for your work....

kakopappa, 2016-10-04:
Thanks :)

You have to change the "persistent_uuid " and "device_name"  to something else if you want to run multiple devices on the network.

richms, 2016-10-28:
It looked in the sketch to me that it was making the persistent uuid from serial or something, is there still a need to change something on that line or is it generating it automatically now?

kakopappa, 2016-11-15:
Hi,

To run multiple instances of ESP switches:

1. persistent_uuid is [generated](https://github.com/kakopappa/arduino-esp8266-alexa-wemo-switch/blob/master/sinric.ino#L148). 

However [Serial No](https://github.com/kakopappa/arduino-esp8266-alexa-wemo-switch/blob/master/sinric.ino#L237) is not so you may need to change this. (Not sure)

another option is to flash micropython and running fauxmo (haven't tested it)

https://github.com/makermusings/fauxmo
https://github.com/micropython/micropython/tree/master/esp8266

joeman2116, 2016-11-19:
I love this program!! Works great! A brilliant piece of software.....

I currently run 2 Sonoff modules, 2 ESP8266-07 and 2 nodemcu modules all loaded with Aruna Tennakoon software.
They all work perfectly  and online all at the same time.
The only changes I did were to change the ** device name for each of the modules** and I also changed the ouput relay pin to work with whatever device I was using....

Tell Alexa to discover and they show up and work
flawlessy....

THANKS!
Joe

kakopappa, 2016-11-26:
I have created a new project to support multiple swictes 
https://github.com/kakopappa/arduino-esp8266-alexa-multiple-wemo-switch

harbirjaswal, 2019-02-26:
multiple switches can be searched with alexa but pin number is not defined...Can you please give an example with pin number...