kakopappa/arduino-esp8266-alexa-wemo-switch #4: Control Multiple GPIOs on ESP-12

jemarr, 2016-10-13:
This is an amazing project, you make my dreams true!!!

I've several ESP8266 (ESP-12E) boards controlling a lot of relays that turn on/off my home lamps...and sensors that reads luminosity. Each writen with his own arduino sketch.

Now it's the time to buy an Amazon Echo and your project could help me a lot.

I use several GPIOs on each ESP-12E board, so, is it possible to modify your project to make the Echo detect multiple devices instead of just one ???

Thanks for your help.

kakopappa, 2016-10-13:
Thanks :)

I think it is possible to emulate more than one wemos switch. Take a look at fauxmo project. 

[https://github.com/makermusings/fauxmo/blob/master/fauxmo.py#L407](https://github.com/makermusings/fauxmo/blob/master/fauxmo.py#L407)

jemarr, 2016-10-13:
Thanks Aruna. I will try to understant Fauxmo python code, I'm not too expert in it !!!

Have you ported Fauxmo python code to Arduino IDE ?

kakopappa, 2016-11-25:
Hi 

I have started working on multiple-wemos-switch based on fauxmo. 

https://github.com/kakopappa/arduino-esp8266-alexa-multiple-wemo-switch

I have not tested it yet. just finish coding.

jemarr, 2016-11-25:
Great, good job.

Thanks

Jesus

kakopappa, 2016-11-26:
Fixed. Now you can emulate more than 1 switch