kakopappa/arduino-esp8266-alexa-wemo-switch #6: Sonoff switch

shenery, 2016-11-03:
Do you think this code could modified to work with the sonoff switch. http://blog.squix.org/2016/01/esp8266-products-hitting-market.html

It has 1M of flash and uses esp8266

shenery, 2016-11-09:
Or do you think it will just work? I'm getting one to test in the next couple of weeks.

joeman2116, 2016-11-14:
It works fine on a sonoff. I just used the arduino ide software to program it - I used the nodemcu.9v on the arduino type - I use them with the alexa echo dot

Great pieces of software...

Joe

shenery, 2016-11-14:
I received 4 today and had a quick go. Using arduino I had it on generic esp8266 board and 1m flash. I changed the relay pin to pin 12 in the code and was getting error esp com mem failed or something like that. Was the pin number all you changed? And do I have to change the board type to nodemcu 9v and that will be me sorted? Really excited about this. An Alexa switch for £5! :-)

shenery, 2016-11-14:
I used a fdti 3.3v

joeman2116, 2016-11-14:
I also use a ftdi 3.3v as the programmer.
I tried type as a generic 1meg flash but got reboot problems.
So I used the nodemcu.9 choice.

Change the ssid
Change password
Change device name to (your choice)

Pins
I use d6 for the relay and d7 for the led.(optional)

Joe

joeman2116, 2016-11-14:
The esp mem code error -likely means your device is not in flash mode

TO flash:

Press and hold the button, do not release: while holding it down, plug in
the programmer, then release button.

Joe

shenery, 2016-11-14:
Ok thanks for the advice. I will give it a try today.

Cheers
Steven

shenery, 2016-11-15:
I tried nodemcu 0.9 esp-12. 1m spiffs and I'm still getting error mem esp com failed. Also d6 doesn't work for me. Isn't the relay gpio 12? It doesn't even compile when I have it as d6 but when I change to 12 it compiles fine.

joeman2116, 2016-11-15:
D6 not d6 - my mistake  gpio12 is D6

Ok, mem esp error - well that is a serial comm problem which means  unit is
not in pgm mode or your wire of the tx and rx pins on your ftdi may not be
correct to the sonoff.

I leave the nodemcu .9 at 3m spiffs but 1meg is fine... The  error your
displaying is not related to those selections.

It is a communication issue related to wiring and or not in correct mode to
flash.

Joe

shenery, 2016-11-17:
Ok so I've been spending a bit more time on this. I have all your settings loaded. I have a fT232RL ftdi like this on http://m.ebay.co.uk/itm/Latest-Version-FT232BL-FTDI-USB-to-TTL-Serial-Adapter-for-Arduino-UK-Seller-/181797979805?nav=SEARCH connected to the sonoff. All wires connected and have been checked numerous times. I have it plugged into my tablet via USB otg and I have also tried through a powered USB hub and I am still getting the espcomm mem failed error. I occasionally get a java lang nullpointerexception error as well. Do you think it could be a power supply issue? Kind regards
Steven

shenery, 2016-11-17:
Also I press the button on the sonos and keep holding it until I plug it in to the tablet and there is no blinking led. I assume that's me in flash mode?

joeman2116, 2016-11-17:
It still appears your system is not in flash mode.
Your steps look fine.
1. Make sure you have selected the correct serial port.. Check windows
system manager under com ports.
2 Make sure the serial monitor is not open.

Joe

joeman2116, 2016-11-17:
The powersupply will not usually be a problem when doing a flash
It mostly acts up when the esp8266 is running and using the sotware and
wifi.

For your programmer,  tx am rx are the typical error area. If all fails
revese the tx and rx wires
Programming note:
Vcc 3.3v
Gnd
Tx
Rx
Are the only wires required.

Joe

shenery, 2016-11-18:
I've tried all of that. Jumper is on the 3.3v.  Vcc is the 3.3v. Checked in device manager which com port is recognised when I connect. Serial monitor window not opened. I saw on this forum that some people were getting the same error and fixed it with clearing the flash with esptool before flashing the new firmware? Do you have any experience with this? I can't see esptool in the programmer of arduino ide.

shenery, 2016-11-18:
Sorry forgot to post the forum. http://tech.scargill.net/itead-slampher-and-sonoff/

joeman2116, 2016-11-18:
Well all sounds good
I have used esptool to clear mem. Esptool is another program. Just google
and download

Sometimes an esp wont work so i use the esptool to refresh it.

Wish you had another regular esp nodemcu to see if you could flash it.
Since they have the programmer builtin, this would establish your serial
programmer is good.
Have you ever successfully flashed any esp8266?

shenery, 2016-11-24:
ok so I got a bit further today. I got this.
83936 bytes from C:\Users\Shenery\AppData\Local\Temp\arduino_build_786479/sinric.ino.bin to flash at 0x00000000
	erasing flash
	size: 045520 address: 000000
	first_sector_index: 0
	total_sector_count: 70
	head_sector_count: 16
	adjusted_sector_count: 54
	erase_size: 036000
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	setting serial port timeouts to 15000 ms
	setting serial port timeouts to 1000 ms
	espcomm_send_command: receiving 2 bytes of data
	writing flash
................................................................................ [ 28% ]
................................................................................ [ 57% ]
................................................................................ [ 86% ]
......................................                                           [ 100% ]
starting app without reboot
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
closing bootloader
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
I plugged in a lamp and I get nothing though. 

when I bring up serial monitor it says f5, fo, f5 f0 and there is no indication that the esp is connected to my network. when scan my network there is no esp device

joeman2116, 2016-11-24:
Was that done with esptool?

Try running it(espool) and writing a blank bin at 0000
Do a google and you should find one
If it writes ok then disconnect and reboot
Then run up the arduino ide program and upload the code for the esp..
Make sure you put it in the flash mode
Joe

If you had teamviewer I could look at the screen etc and maybe help more..

shenery, 2016-11-24:
I set up esptool but not 100% of the command to use to clear the flash. Tried a few but didn't work. I have a blank.bin file saved in the esptool directory. I have installed team viewer and could maybe try that over the weekend if you have time. If you have a command to use in esptool though I could try that?

Kind regards
Steven

shenery, 2016-11-24:
And when I try the erase flash command in esptool I get esptool error: argument --port/-p: expected one argument

shenery, 2016-11-26:
Ok so I've been having fun getting this going. I found the right command for esptool and have cleared the flash, I was then able to upload the firmware again. I've realised that there is a certain way that you have to wire the lamp in to so that the sonoff is powered via the mains. I scanned devices in alexa and it found my lamp. I asked Alexa to turn on box and she replied ok. But nothing happens the light stays off. I have tested the fuse the bulb and tried in another lamp but it doesn't turn on and I've no idea why.
I'm nearly there but not sure what to do next.

shenery, 2016-11-26:
Scrap that last post. I forgot to change the gpio pin again and now it works. If anyone's having bother esptool helped out a lot. Basically in my experience you have to use esptool to clear flash before you upload this code. Thank you joeman for all your help it works perfect no :-)

joeman2116, 2016-11-26:
Great, glad you kept at it.
Joe

shenery, 2016-11-28:
Yeah it works really good.
My wifis been down all day. What a coincidence eh? Lol. This happening has gave me anurgancy to try and work out how to override the relay with say a hardware switch or maybe you can use the program button?
Has anyone done this that you know of?

Kind
Regards

kakopappa, 2016-11-29:
You can use a push button.
https://www.arduino.cc/en/tutorial/switch

when HIGH call turnOnRelay();
when LOW call turnOffRelay();

shenery, 2016-11-29:
I'm not very good with adding code so not sure where to put it in the script. Where would it go? And would I be able to use the program button on gpio 0 for this? If not would the switch go at the mains side or the lamp side?

Kind regards
Steven

joeman2116, 2016-11-29:
The esp com failure means it is not likely in flash mode
Make sure u hold down the button an dont let go until a few seconds after
you plug in the programmer.

Joe

shenery, 2016-11-29:
I have found this piece of code from sonoff boilerplate firmware. Looks like the code to use the button on gpio 0 to switch the relay.
Can I just add this somewhere in the code?
 //delay(200);
  //Serial.println(digitalRead(SONOFF_BUTTON));
  switch (cmd) {
    case CMD_WAIT:
      break;
    case CMD_BUTTON_CHANGE:
      int currentState = digitalRead(SONOFF_BUTTON);
      if (currentState != buttonState) {
        if (buttonState == LOW && currentState == HIGH) {
          long duration = millis() - startPress;
          if (duration < 1000) {
            Serial.println("short press - toggle relay");
            toggle();
          } else if (duration < 5000) {
            Serial.println("medium press - reset");
            restart();
          } else if (duration < 60000) {
            Serial.println("long press - reset settings");
            reset();
          }
        } else if (buttonState == HIGH && currentState == LOW) {
          startPress = millis();
        }
        buttonState = currentState;
      }
      break;
  }

}

https://github.com/tzapu/SonoffBoilerplate/blob/master/SonoffBoilerplate.ino

terrag42, 2016-11-29:
I tried out this code last night. It works fine for tactile switches. I tried it with one of those cheapo switches with a spring. It worked but there was a lot of rapid turn on / off because there is no debounce in the code.

burhantetik, 2016-12-09:
Hello,

I need the compiled bin file of sinric.ino. Can please someone send it to my mail? burhant@gmail.com

Shooter66six, 2016-12-14:
So where dose this code go in order to make the button work?

//delay(200);
//Serial.println(digitalRead(SONOFF_BUTTON));
switch (cmd) {
case CMD_WAIT:
break;
case CMD_BUTTON_CHANGE:
int currentState = digitalRead(SONOFF_BUTTON);
if (currentState != buttonState) {
if (buttonState == LOW && currentState == HIGH) {
long duration = millis() - startPress;
if (duration < 1000) {
Serial.println("short press - toggle relay");
toggle();
} else if (duration < 5000) {
Serial.println("medium press - reset");
restart();
} else if (duration < 60000) {
Serial.println("long press - reset settings");
reset();
}
} else if (buttonState == HIGH && currentState == LOW) {
startPress = millis();
}
buttonState = currentState;
}
break;
}

}

shenery, 2016-12-15:
I'm not sure either I haven't had time to try and get it working. Not sure if @joeman2116 might have it working. I'm sure he said some thing about having the led working but not sure about the button.

Shooter66six, 2016-12-15:
If you or any1 else manages to sort it please let me know.  Thank you

joeman2116, 2016-12-15:
Been finishing a 4 channel multiswitch with the new code, been busy.. But it all working now.

As for adding switch control, 
Have not looked at the best place for the code
But expect one could make it a declared as a void routine and then call the routine inside void setup.
I will try to test later

Joe

Shooter66six, 2016-12-16:
Any luck Joe?

shenery, 2016-12-27:
He Joe. Hope you've had a good Christmas. Did you manage to get any further with the switch on the sonoff. I have worked out a way to attach a switch to the relay but it's a bit clumsy. Utalising the button on the sonoff would be a lot better.

joeman2116, 2016-12-27:
Yes, Christmas was good and a very busy time.
I have also been working with another chap helping to test some new code
he  has developed as an enhancement to the Weemo emulator software.

The addons will allow the end user to configure the wifi and devices names
from its own builtin server.  Will work with most all esp based modules.

Once it is tested and released, i will look to see if there is a simple
routine available to add to allow a sonof switch to be used if available.

Merry Christmas 🎅 from Canada

Joe

MacGyver1rjq104, 2016-12-28:
Does anyone know if this code will work with the sonoff dual switch.  It looks like the dual has two places to upload code.  I was wondering if this is just two device put on one board.  Does anyone know?

yaydogan, 2016-12-30:
@shenery   How did you clear the flash using esptool? I am having similar issues you had before clearing flash. 

Thanks in advance.

shenery, 2016-12-30:
@yaydogan if you have python set up in Windows I used this code to get it to working changing the port number. CD in to the location where ESP tool is and enter this. python esptool.py -p com6 erase_flash

joeman2116, 2016-12-30:
I put 4 blank bins. Tell esptool to to write
It will go thru each address clearing the esp8266.  U can do a Google for
cleaning esp8266 with blank bins.
Joe

TCRAKN, 2017-01-04:
@shenery

I've modified the sketch to add a momentary switch to D2, and to send the state to the serial port.  The code is rudimentary, but it work and is stable.

The only problem I have is that the initial connection to WiFi is not reliable - it times-out - so then have to manually reset.  I've increased the loop count from 10 to 20, to try to give it more time to connect.

I don't know how to save this is a proper GitHub repository.  If it helps, sketch is below...

Best
Abdul-Khaliq
_____

// Modified 4 Jan 2016 AKN
// To include a momentary switch to switch relay on and off

#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <WiFiUdp.h>
#include <functional>

void prepareIds();
boolean connectWifi();
boolean connectUDP();
void startHttpServer();
void turnOnRelay();
void turnOffRelay();

const char* ssid = "SSID";
const char* password = "PASSWORD";

unsigned int localPort = 1900;      // local port to listen on

WiFiUDP UDP;
boolean udpConnected = false;
IPAddress ipMulti(239, 255, 255, 250);
unsigned int portMulti = 1900;      // local port to listen on

ESP8266WebServer HTTP(80);
 
boolean wifiConnected = false;

char packetBuffer[UDP_TX_PACKET_MAX_SIZE]; //buffer to hold incoming packet,

String serial;
String persistent_uuid;
String device_name;

const int relayPin = D1;
const int switchPin = D2;

int switchState = 0;

boolean cannotConnectToWifi = false;

void setup() {
  Serial.begin(115200);

  // Setup Relay
  pinMode(relayPin, OUTPUT);

  // Setup Switch
  pinMode(switchPin, INPUT);  
  
  prepareIds();

  delay(3000);
  // Initialise wifi connection
  wifiConnected = connectWifi();

  // only proceed if wifi connection successful
[... 273 lines cut ...]

boolean connectUDP(){
  boolean state = false;
  
  Serial.println("");
  Serial.println("Connecting to UDP");
  
  if(UDP.beginMulticast(WiFi.localIP(), ipMulti, portMulti)) {
    Serial.println("Connection successful");
    state = true;
  }
  else{
    Serial.println("Connection failed");
  }
  
  return state;
}

void turnOnRelay() {
 digitalWrite(relayPin, HIGH); // turn on relay with voltage HIGH 
}

void turnOffRelay() {
  digitalWrite(relayPin, LOW);  // turn off relay with voltage LOW
}

Shooter66six, 2017-01-16:
thank you TCRAKN  for looking at the code, but im having problems with it.
it will not verify.
Arduino: 1.6.13 (Windows 10), Board: "NodeMCU 0.9 (ESP-12 Module), 80 MHz, 115200, 4M (1M SPIFFS)"

new_sonoff:11: error: #include expects "FILENAME" or <FILENAME>

 #include 

          ^

exit status 1
#include expects "FILENAME" or <FILENAME>

can you help me out please.

TCRAKN, 2017-01-17:
Hi Shooter66six,

I’ve been compiling successfully with Arduino IDE 1.6.12 (Mac OS X):
Board: NodeMCU 1.0 (ESP-12E Module) 80 MHz, 115200, 4M (3M SPIFFS)

There are a couple of warnings relating to line 163, but it compiles and uploads successfully.

I have since modified the sketch for the Sonoff to take into account Sonoff pin designations for the Relay, Switch and Green LED, but have yet to test it.  Once I locate my USB-Serial module, I’ll get right to it.  This version also compiles successfully.

Having looked closer at the error you’re experiencing suggests that I may have accidentally deleted part of the #include statement!  It should read:

#include <functional>

Though, omitting this line completely should pose no problems, as this library is part of the ESP8266 Aurduino core toolchain.

My apologies to everyone who might be pulling their hair out!

Abdul-Khaliq

Shooter66six, 2017-01-18:
i added <functional> then got this
Arduino: 1.6.13 (Windows 10), Board: "NodeMCU 0.9 (ESP-12 Module), 80 MHz, 115200, 4M (1M SPIFFS)"

C:\Users\Scott\Documents\Arduino\sketch_with_switch\sketch_with_switch.ino: In function 'void startHttpServer()':

sketch_with_switch:197: error: expected '}' before ';' token

 Serial.println("Got Request index.html ...\n");

                                               ^

sketch_with_switch:197: error: expected ')' before ';' token

C:\Users\Scott\Documents\Arduino\sketch_with_switch\sketch_with_switch.ino: At global scope:

sketch_with_switch:199: error: expected unqualified-id before ')' token

 });

  ^

sketch_with_switch:201: error: 'HTTP' does not name a type

 HTTP.on("/upnp/control/basicevent1", HTTP_POST, []() {

 ^

sketch_with_switch:225: error: expected unqualified-id before ')' token

 });

  ^

sketch_with_switch:227: error: 'HTTP' does not name a type

 HTTP.on("/eventservice.xml", HTTP_GET, [](){

 ^

sketch_with_switch:258: error: expected unqualified-id before ')' token

 });

  ^

sketch_with_switch:260: error: 'HTTP' does not name a type

 HTTP.on("/setup.xml", HTTP_GET, [](){

 ^

sketch_with_switch:295: error: expected unqualified-id before ')' token

 });

  ^

sketch_with_switch:297: error: 'HTTP' does not name a type

 HTTP.begin();  

 ^

sketch_with_switch:298: error: 'Serial' does not name a type

 Serial.println("HTTP Server started ..");

 ^

sketch_with_switch:300: error: expected declaration before '}' token

 }

 ^

exit status 1
expected '}' before ';' token

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

TCRAKN, 2017-01-18:
@Shooter66six 

I'm beginning to think that that wasn't the only change the Github editor has made!

You can download my latest sketch from here:
https://expirebox.com/download/dbeb85ad9b91ebd846e797e59efc2fea.html

It'll expire in a couple of days.

Best
Abdul-Khaliq

Shooter66six, 2017-01-18:
that was the problem. the new sketch flashed on first time.
however asking alexa or pressing the switch only momentarily turns off the power, it flicks back on after a second. is that how you did it or is that a fault? also could you make it so it says off insted of flicking off then back on.

also thank you for this Abdul-Khaliq your a top man.

TCRAKN, 2017-01-18:
@Shooter66six 

It's working fine on my Wemos D1 setup, and the Sonoff isn't much different.

I don't have my test units with me at the moment, so cannot re-test.  

Can you launch the serial monitor from the Arduino IDE to see what's being reported, and post here?

I've moved around a couple of variable settings for switchState to how it is in my Wemos Sketch, so it may be worth testing this version...
https://expirebox.com/download/bfdadd8d4ac26a3254c6472d1c889b07.html

Best
Abdul-Khaliq

MacGyver1rjq104, 2017-01-19:
I was wondering.  Is there a way to control this with a wifi app like Belkin Wemos do so I can also set timers?  and.... Is there a way to program  it  so you can change the Wifi password and SSID by wireless like the Sonoff does before we changed the code?  Basically if it were possible, it would be nice if this would emulate the wemo (wemo clone).  BTW.... Thanks for your posting and help.  This is really awesome!!!!!

TCRAKN, 2017-01-19:
@MacGyver1rjq104 

@joeman2116 mentions further up this thread that a more configurable version is in the works.  Any updates?  https://github.com/kakopappa/arduino-esp8266-alexa-wemo-switch/issues/6#issuecomment-269377620 

As for an App, I'm not sure if the existing Belkin apps for iOS or Android will recognise broadcasts from this firmware.  Can someone report?  Maybe all you'd need to issue is a properly framed xml stream to a device via a URL.  Perhaps something here might help: https://github.com/issackelly/wemo 

Regards
Abdul-Khaliq

MacGyver1rjq104, 2017-01-19:
@TCRAKN 

Thanks... The https://bitbucket.org/xoseperez/espurna site was the first one I looked at.  I didn't understand his methods because he insists on using other programs NOT Arduino which is the one I am familiar with.  I spent days trying to figure out what he was doing then I found your version.   It's much more direct in explaining the process.  Thank you.  I am new to C++ so I am struggling to learn.  I can usually figure out how to modify bits and pieces but I want to get to a level where I can program from scratch.  The best way to learn is to dive in and get familiar with the code and how it works.  I am looking forward to seeing the development from @joeman2116

TCRAKN, 2017-01-19:
@MacGyver1rjq104 

There is a very good Wiki linked to it here:  https://bitbucket.org/xoseperez/espurna/wiki/ArduinoIDE.md

It's somewhat thorough, but you're right - there is too much to follow.  I will give it a go, but that might be some time away. 

I hope my modified version works out for you in the meantime.  I can't take any credit to the real work that's behind it working. Yes, I'm looking forward to seeing what @joeman2116 is working on too!

@joeman2116 any teasers?

Best wishes
Abdul-Khaliq

Shooter66six, 2017-01-19:
the new sketch works very well as it should. thank you for your hard work Abdul

shenery, 2017-01-19:
@Shooter66six So we have a working switch on the sonoff then? Does the black button on the sonoff turn the relay on and off bypassing the WiFi?

TCRAKN, 2017-01-20:
@shenery 

The sketch posted here:https://github.com/kakopappa/arduino-esp8266-alexa-wemo-switch/issues/6#issuecomment-273629923 does this:

1. On power up, the relay is on, and the green LED is also on.
2. Invoking Alexa to change the state will also change the state on the LED as per on/off request.
3. Pressing and releasing the button will change the current state of the LED and relay.
4. Only on the release of the button will the change take effect.
5. WiFi connectivity isn't necessary for the button to work.

The link above will expire tomorrow, I think.  I'm not sure how to add/fork the original repository into a new one containing this sketch.  So, if anyone wants it after the link expires, just drop me a line here.

Best
Abdul-Khaliq

MacGyver1rjq104, 2017-01-21:
Everything is looking great... Quick question....  Do you have a solution for the Sonoff Dual relay?  I tried your WeMo multi relay firmware.  It finds the device and Identifies two switches on alexa but will not control the relays.  I am assuming that I need to identify the D6 pin and what ever the 2nd relay is tied to but I am still learning how to do things like this.  Any help, insight is appreciated!!!

shenery, 2017-01-21:
@TCRAKN It sounds really good I will try it tomorrow. I have the download saved. On an unrelated post. Do you know if this code could be manipulated by openhab in some way?

Shooter66six, 2017-01-22:
Could the sketch be made so it starts as off and turns on momentarily on a click of the switch or alexa command? This could then be flashed onto the 12v sonoff and used to turn on your pc

TCRAKN, 2017-01-22:
@MacGyver1rjq104 
All my current requirements are for single-relay Sonoffs, so I've had no ned to look into the multi-relay code.  I must add, the code is by @kakopappa, so it might be worth dropping Aruna a line.

Sorry I cant help any further.
Abdul-Khaliq

TCRAKN, 2017-01-22:
@Shooter66six 

Yes, just remove the line _turnOnRelay();_ at the end of the _void setup() {_ function and change _int switchState = 1;_ to _int switchState = 0;_ just before  _void setup() {_

Best
Abdul-Khaliq

[32 more replies]