kakopappa/arduino-esp8266-alexa-wemo-switch #7: alexa not discovering device

parus21, 2016-11-10:
thanks for making such a cool project searched all over for a way to control a relay through alexa. now if only i can get it to work.  i run the code and open serial monitor. it says it connects to my wifi network and that http server started but when i try to discover new devices with alexa it doesnt show up? any help would be appreciated.

kakopappa, 2016-11-12:
1. Make sure ports or udp packets are not blocked from router.

2. When you ask alexa to discover devices, do you see these printed in the serial window ? normally alexa does a udp broadcast when you request to discover new devices

########## Responding to setup.xml ... ########

or 

########## Responding to eventservice.xml ... ########

in the serial monitor?

michael040990, 2016-11-14:
I have also tried to set up this device and do not get a response  all devices are on the same network  UDP seems to be working on the network as I am getting a request showing up from the router on the serial port   However nothing from Alexa  i'm using Alexa  on the Amazon fire tv using a ESP-01  if I navigate to the ESP's homepage  Aka index.htnl  I get hello world so I know it is connected

joeman2116, 2016-11-14:
I am using the software on several nodemcu 8266 -12f modules and it works great. 
Also reprogrammed the Sonoff modules with this code. 
I use the arduino ide software to flash the new code.  I choose the nodemcu.9v type 

Changed the ssid and password and gave each one a new device name, reboot and
 Told Alexa Echo dot to discover and all appeared.... wow! great stuff....

Thanks for Super Piece of Software!!
Joe

parus21, 2016-11-17:
just looking back at the boards i ordered and realized that i ordered the d1 mini pro which looks like a different board. could this be causing the issue?

parus21, 2016-11-17:
not seeing anything when i make a request from alexa in the serial monitor. just pauses at http server started.

joeman2116, 2016-11-17:
I looked at the mini pro and it looks fine. It has more mem than most.
Should work ok.
I use frpm esp01 to 12f and nodemcu modules and all work ok.
Is your power source powerfull enough?
Joe

AndrewsHappy, 2016-12-18:
It is beautiful code, and I am glad that someone has got it to work, but afraid I am not one of them. I know it is getting onto the router, as i can see it, I can also ping it. Just Alexa will not discover it. Granted that I have a professional meru wifi system, which has lots of settings that might stop the UDP etc... but also tried it on my back up router that EE provides and still nothing... You can't get the WeMo sockets for love or money (and a lot of money at that) to see if your network is the problem of if there is a bit of code that just doesnt like me!  So close yet so far!

joeman2116, 2016-12-18:
Are you referring to the single device or multiple device version?

When you run the serial monitor and reset your esp do you see it attempting
to connect?

If so then  tell Alexa to discover devices

You should see the requests ack coming from Alexa
and when you tell Alexa to turn on the device you should see it print
across your screen?

Joe

AndrewsHappy, 2016-12-18:
I must confess that I am not as clever as I might think I am and quite new to it all.  I am programming the Sonoff and only got as far as looking at the single device.  I do not know how to use the Serial Monitor to get anything useful out of it, so I do not even know if Alex is doing anythin....HOLY SMOKE!! Press the button.... I didn't even realise I needed to push the button.... DHO! 

OK, I now have a WeMo Switch showing up on my Alexa.... Joe, you are my hero!!! Thank you....

joeman2116, 2016-12-18:
Glad I could help..  I have several sonoff modules setup up as singles
devices on Alexa -  they work very well.

The serial monitor is an option located under one of the arduino ide tabs.
When you reset the module, it will display the actions  (baud rate at
bottom  should display 9600 or you will get weird character prints
If all is well you will see things like:

Such as connecting
Connected to wifi  xxx
IP
It helps a lot when trying to see if all is working properly and where the
failure might be etc...

The muliple version is also very cool  when you are using regular esp8266
modules as you can have one esp8266 control up 2 14 devices..
I have a few boxes with 1 esp8266 a 4 channel relay and Alexa displays it
as 4 devices each with their own name,,

Great piece of software....

Joe

AndrewsHappy, 2016-12-18:
Slow down! So with the Sonoff can you get the relay and green LED to work together, at the moment, it seems to only be "Alexa turn the light on" to turn on the light, then "Alexa turn the light on" to turn it off!!  I think I need to study this code more... but at least I can now go to bed with some sense of achievement

joeman2116, 2016-12-18:
Yes,
You just need to make sure the declared pins are correct.

1. One for the relay.

2. One for the LED - which must be added.

Then you would need to add the same code for the led.

joe

AndrewsHappy, 2016-12-18:
Yes, it isn't as easy as I thought

void turnOnRelay() {
 digitalWrite(relayPin, HIGH); // turn on relay with voltage HIGH
 digitalWrite(ledPin, HIGH); // turn on the green LED 
}

giving the value of 13 for the relay and 12 for the relay, but all it seems to do is to turn on the relay and not the LED. 

Oh, thanks again, got the serial monitor working.... having fun now,..... shame it is 1am

joeman2116, 2016-12-18:
const int relayPin = D1;
const int ledPin = D7;

void turnOnRelay() {
 digitalWrite(relayPin, HIGH); // turn on relay with voltage HIGH
 digitalWrite(ledPin, LOW); //GREEN LED
}

void turnOffRelay() {
  digitalWrite(relayPin, LOW);  // turn off relay with voltage LOW
  digitalWrite(ledPin, HIGH); //GREEN LED OFF
}

This is what I have on my Sonoff.... to control led and relay

Ltr.
.
Joe- Canada

AndrewsHappy, 2016-12-18:
One step behind you!
Right, off to bed
Night

Andrew (Hampshire, England)

AndrewsHappy, 2016-12-19:
Hi Joe

do you find that you have to keep turning the sonoff off and then on after an hour to get it to register back onto the wifi ? Mine is falling asleep and Alexa says it is off line.

Andrew

joeman2116, 2016-12-19:
Nope
I have several sonof running 24/7 for more than a month so far they work
100 percent

AndrewsHappy, 2016-12-19:
Now I have given it a static in, seems to be behaving.

Have you had any experience programming the Linknode R4, seems kakopappa has written some code, and although it will flash, does not do anything according to the serial monitor, yet I can put the Sonoff onto it and get it to join the net...

Tomorrow I am getting a Broadlink RM Pro, which they say can also be made to talk to Alexa...

@ndrew

-

joeman2116, 2016-12-19:
Would be interesting to see what code you used for static ip.

Have not looked at the link node R4 to see what it is supposed to do.

Joe

AndrewsHappy, 2016-12-20:
I just went to the router and found the MAC code of the Sonoff and assigned it a static IP. I have lots of IP cameras, VoIP, AV's and now this lot... so I like to try and keep track of my ip addresses by giving them all static ip's

The Link Node is just a 8266 with 4 relays on it that you can control like the Sonoff, as well as flash code onto it...

joeman2116, 2016-12-20:
> switch/issues/7#issuecomment-267797037
> AWDpY25d4o3XYVF346stZ22g7Os3oWYlks5rJIdugaJpZM4KvPK8
> switch/issues/7#issuecomment-267797296
> PhVcnbqIhnz7sSGKb2iks5rJIlEgaJpZM4KvPK8
> switch/issues/7#issuecomment-268047872
> m92CAmaijZICiOcnN2KXLcks5rJtQagaJpZM4KvPK8
> ARQwPVHAiQz3hPrCtyRsRGCLRN_GWYfqks5rJw2_gaJpZM4KvPK8
> kakopappa/arduino-esp8266-alexa-wemo-switch/issues/7#

Thanks for info on the linknode  - ordered 2

The broadlink rm pro looks interesting as well

joe

AndrewsHappy, 2016-12-20:
Let me know if you get either working with Alexa, seems only one person has so far.

@ndrew

-

joeman2116, 2016-12-21:
Got the linknodes r4 in today and 10 minutes later Alexa is working fine
with the nodes 4 relays.

I just added a 5v +center powersupply
Connected 3 wires from my uart (programmer) tx rx  and gnd (3vmode)
Moved the jumper to the UART PROGRAM position

Then i ran esptool to write blank bins  - clears memory - probably not
required but have found it clears any issues - so I do it first.

Then I ran the code found at " The makers workbench"   How to use Amazon
Alexa to control the Linknode R4.
I changed the ssid and pw.

Compiled and uploaded use the Arduinio IDE software

Generic 8266
QIO
2meg /1 meg spfss

makersite
<https://themakersworkbench.com/tutorial/how-use-amazon-alexa-control-linknode-r4-esp8266-4-channel-relay-board>
is
here..

Ran the node thru its paces - on, off / combinations etc - works great.
Now will exercise it during the eve and overnight to see if there are any
issues.

So far ---- cool....

Joe

AndrewsHappy, 2016-12-21:
Wow, in 10 minutes! Took me a few days ! But now I know what I'm doing...

I got my broadlink rm pro, as I have a few infrared devices I'd like to get Alexa to control, but seems there is a better way using a Logitech Harmony hub.... but not keen on having to use the cloud to connect to Alexa...

@ndrew

<https://themakersworkbench.com/tutorial/how-use-amazon-alexa-control-linknode-r4-esp8266-4-channel-relay-board>

-

solrac21, 2016-12-21:
Hello everyone, i am having some isssues...I can flash the Sonoff and i get this in the serial monitor:
.
Connected to HOME-1307-2.4
IP address: 10.0.0.229

Connecting to UDP
Connection successful
HTTP Server started ..

but no matter what i do, i cannot get alexa to see it.  I also do not see the request from Alexa on the serial monitor.  I have tried diffrent UDP ports but nothing....

AndrewsHappy, 2016-12-21:
Are you pressing the button on the Sonoff once you have flashed it and the Sonoff has logged onto the network and asked Alexa to find new devices?

joeman2116, 2016-12-21:
Curious how the Broadlink Rm Pro works out with Alexa. It has the most
device connectivity.

Cheaper than Harmony but Harmony more polished interface.

Thanks for the linknode info, I like them..

Joe

https://themakersworkbench.com/tutorial/how-use-amazon-alexa-control-linknode-r4-esp8266-4-channel-relay-board
is
here..

Ran the node thru its paces - on, off / combinations etc - works great.
Now will exercise it during the eve and overnight to see if there are any
issues.

So far ---- cool....

Joe

> Let me know if you get either working with Alexa, seems only one person
> has so far.
>
> @ndrew
>
> Thanks for info on the linknode - ordered 2
>
> The broadlink rm pro looks interesting as well
>
> joe
>
> > I just went to the router and found the MAC code of the Sonoff and
> > assigned it a static IP. I have lots of IP cameras, VoIP, AV's and now
> this
> > lot... so I like to try and keep track of my ip addresses by giving them
> > all static ip's
> >
> >
> > The Link Node is just a 8266 with 4 relays on it that you can control
> like
> > the Sonoff, as well as flash code onto it...
notifications@github.com><mailto:notifications@github.
> com>>
> > Sent: 19 December 2016 23:51
> >
> > To: kakopappa/arduino-esp8266-alexa-wemo-switch
> > Cc: AndrewsHappy; Comment
> > discovering device (#7)
> >
> > Would be interesting to see what code you used for static ip.
> >
> > Have not looked at the link node R4 to see what it is supposed to do.
> >
> > Joe
<mailto:notifications@github.com><
> mailto:notifications@github.com>>
> >
> > > Now I have given it a static in, seems to be behaving.
> > >
> > > Have you had any experience programming the Linknode R4, seems
> kakopappa
> > > has written some code, and although it will flash, does not do
anything
> > > according to the serial monitor, yet I can put the Sonoff onto it and
> get
> > > it to join the net...
> > >
> > > Tomorrow I am getting a Broadlink RM Pro, which they say can also be
> made
> > > to talk to Alexa...
> > >
> > > @ndrew
> > >
> > > Nope
> > > I have several sonof running 24/7 for more than a month so far they
> work
> > > 100 percent
<mailto:notifications@github.com><
> mailto:notifications@github.com>
> > >
> > > > Hi Joe
> > > >
> > > >
> > > > do you find that you have to keep turning the sonoff off and then on
> > > after
> > > > an hour to get it to register back onto the wifi ? Mine is falling
> > asleep
> > > > and Alexa says it is off line.
> > > >
> > > >
> > > > Andrew
notifications@github.com>>>
> > > > Sent: 18 December 2016 01:15
> > > > To: kakopappa/arduino-esp8266-alexa-wemo-switch
> > > > Cc: AndrewsHappy; Comment
> > > > discovering device (#7)
> > > >
> > > > const int relayPin = D1;
> > > > const int ledPin = D7;
> > > >
> > > >
> > > > void turnOnRelay() {
> > > > digitalWrite(relayPin, HIGH); // turn on relay with voltage HIGH
> > > > digitalWrite(ledPin, LOW); //GREEN LED
> > > > }
> > > >
> > > > void turnOffRelay() {
> > > > digitalWrite(relayPin, LOW); // turn off relay with voltage LOW
> > > > digitalWrite(ledPin, HIGH); //GREEN LED OFF
> > > > }
> > > >
> > > >
> > > > This is what I have on my Sonoff.... to control led and relay
> > > >
> > > > Ltr.
> > > > Joe- Canada
> > > >
> > > >
> > > > On Sat, Dec 17, 2016 at 9:07 PM, AndrewsHappy <
notifications@github.com>
> > > <mailto:notifications@github.com>>
> > > >
> > > > > Yes, it isn't as easy as I thought
> > > > >
> > > > > void turnOnRelay() {
> > > > > digitalWrite(relayPin, HIGH); // turn on relay with voltage HIGH
> > > > > digitalWrite(ledPin, HIGH); // turn on the green LED
> > > > > }
> > > > >
> > > > > giving the value of 13 for the relay and 12 for the relay, but all
> it
> > > > > seems to do is to turn on the relay and not the LED.
> > > > >
> > > > > Oh, thanks again, got the serial monitor working.... having fun
> > > now,.....
> > > > > shame it is 1am
> > > > >
> > > > > -
> > > > > <
> > > >
> > > https://github.com/kakopappa/arduino-esp8266-alexa-wemo-
> > switch/issues/7#issuecomment-267797037
> > > > >,
> > > > > <
> > > >
> > AWDpY25d4o3XYVF346stZ22g7Os3oWYlks5rJIdugaJpZM4KvPK8
> > > > >
> > > > >
> > > >
> > > > -
> > > >
> > > https://github.com/kakopappa/arduino-esp8266-alexa-wemo-
> > switch/issues/7#issuecomment-267797296
> > > >,
> > > >
> > PhVcnbqIhnz7sSGKb2iks5rJIlEgaJpZM4KvPK8
> > > >
> > > > -
> > > > <
> > > https://github.com/kakopappa/arduino-esp8266-alexa-wemo-
> > switch/issues/7#issuecomment-268047872
> > > >,
> > > > <
> > m92CAmaijZICiOcnN2KXLcks5rJtQagaJpZM4KvPK8
> > > >
> > > >
> > >
> > > -
> > > https://github.com/kakopappa/arduino-esp8266-alexa-wemo-
> > switch/issues/7#issuecomment-268104804>,
> > ARQwPVHAiQz3hPrCtyRsRGCLRN_GWYfqks5rJw2_gaJpZM4KvPK8
> > >
> > > -
> > > <https://github.com/kakopappa/arduino-esp8266-alexa-wemo-
> > switch/issues/7#issuecomment-268111322>,
> > AWDpY4eoOOj1HOvidUWFubmEIV40vjDrks5rJxYVgaJpZM4KvPK8>
> > >
> >
> > -
> > kakopappa/arduino-esp8266-alexa-wemo-switch/issues/7#
> > Utks5rJxhlgaJpZM4KvPK8>.
> >
> > -
> > <https://github.com/kakopappa/arduino-esp8266-alexa-wemo-
> switch/issues/7#issuecomment-268114707>,
> AWDpY3S2yBphr0gH7bxZ7KnkfFIqza-4ks5rJxsJgaJpZM4KvPK8>
> >
>
> -
> kakopappa/arduino-esp8266-alexa-wemo-switch/issues/7#
> Q8nvIri3juf3t1ks5rJ00rgaJpZM4KvPK8>.
>
> -
> <
https://github.com/kakopappa/arduino-esp8266-alexa-wemo-switch/issues/7#issuecomment-268170763
>,
> <
https://github.com/

-
https://github.com/kakopappa/arduino-esp8266-alexa-wemo-switch/issues/7#issuecomment-268622873>,

AndrewsHappy, 2016-12-21:
Link nodes have suddenly gone up in price here and hard to get. I've had to order more from the USA.

Think I have to use an Android bridge as well. So far not impressed with the Broadlink, that might be down to poor instructions and I'm doing it wrong

https://themakersworkbench.com/tutorial/how-use-amazon-alexa-control-linknode-r4-esp8266-4-channel-relay-board
is
here..

Ran the node thru its paces - on, off / combinations etc - works great.
Now will exercise it during the eve and overnight to see if there are any
issues.

So far ---- cool....

Joe

> Let me know if you get either working with Alexa, seems only one person
> has so far.
>
> @ndrew
>
> Thanks for info on the linknode - ordered 2
>
> The broadlink rm pro looks interesting as well
>
> joe
>
> > I just went to the router and found the MAC code of the Sonoff and
> > assigned it a static IP. I have lots of IP cameras, VoIP, AV's and now
> this
> > lot... so I like to try and keep track of my ip addresses by giving them
> > all static ip's
> >
> >
> > The Link Node is just a 8266 with 4 relays on it that you can control
> like
> > the Sonoff, as well as flash code onto it...
> >
> >
> > From: joeman2116 <notifications@github.com<mailto:notifications@github.com><mailto:
notifications@github.com<mailto:notifications@github.com>><mailto:notifications@github.
> com>>
> > Sent: 19 December 2016 23:51
> >
> > To: kakopappa/arduino-esp8266-alexa-wemo-switch
> > Cc: AndrewsHappy; Comment
> > discovering device (#7)
> >
> > Would be interesting to see what code you used for static ip.
> >
> > Have not looked at the link node R4 to see what it is supposed to do.
> >
> > Joe
<mailto:notifications@github.com><
> mailto:notifications@github.com>>
> >
> > > Now I have given it a static in, seems to be behaving.
> > >
> > > Have you had any experience programming the Linknode R4, seems
> kakopappa
> > > has written some code, and although it will flash, does not do
anything
> > > according to the serial monitor, yet I can put the Sonoff onto it and
> get
> > > it to join the net...
> > >
> > > Tomorrow I am getting a Broadlink RM Pro, which they say can also be
> made
> > > to talk to Alexa...
> > >
[... 118 lines cut ...]
> > Utks5rJxhlgaJpZM4KvPK8>.
> >
> > -
> > <https://github.com/kakopappa/arduino-esp8266-alexa-wemo-
> switch/issues/7#issuecomment-268114707>,
> AWDpY3S2yBphr0gH7bxZ7KnkfFIqza-4ks5rJxsJgaJpZM4KvPK8>
> >
>
> -
> kakopappa/arduino-esp8266-alexa-wemo-switch/issues/7#
> Q8nvIri3juf3t1ks5rJ00rgaJpZM4KvPK8>.
>
> -
> <
https://github.com/kakopappa/arduino-esp8266-alexa-wemo-switch/issues/7#issuecomment-268170763
>,
> <
https://github.com/

-
https://github.com/kakopappa/arduino-esp8266-alexa-wemo-switch/issues/7#issuecomment-268622873>,

-

-

solrac21, 2016-12-22:
No, i haven't tried the button.  So do i hold the button down? press the button once and scan for devices?

AndrewsHappy, 2016-12-22:
Press the Find devices (or ask her to) then press the button. It should then come up on the app

solrac21, 2016-12-22:
Still nothing,,, i have a fire stick and tried the app but neither one sees it.  Am i supposed to see the requests from the alexa units in the serial monitor?

AndrewsHappy, 2016-12-22:
Soon after you have flashed the sonoff, the serial monitor should then do the following

Connecting to WiFi

Connecting ................
Connected to "Your SSID"
IP address: 192.168.1.10 (or what ever your IP is set up as)

Connecting to UDP
Connection successful
HTTP Server started ..

Received packet of size 173
From 192.168.1.4, port 58406

this will repeat a few times.
Get Alexa to discover

Sending response to 192.168.1.36
Port : 50000
Response sent !
 ########## Responding to setup.xml ... ########

Sending :<?xml version="1.0"?><root><device><deviceType>urn:Belkin:device:controllee:1</deviceType><friendlyName>Test Relay</friendlyName><manufacturer>Belkin International Inc.</manufacturer><modelName>Emulated Socket</modelName><modelNumber>3.1415</modelNumber><UDN>uuid:Socket-1_0-38323636-4558-4dda-9188-cda0e6cbf88a</UDN><serialNumber>221517K0101769</serialNumber><binaryState>0</binaryState><serviceList><service><serviceType>urn:Belkin:service:basicevent:1</serviceType><serviceId>urn:Belkin:serviceId:basicevent1</serviceId><controlURL>/upnp/control/basicevent1</controlURL><eventSubURL>/upnp/event/basicevent1</eventSubURL><SCPDURL>/eventservice.xml</SCPDURL></service></serviceList></device></root>

Received packet of size 173
From 192.168.1.53, port 2048

And now it should show up on your app.

kakopappa, 2016-12-22:
If you are not seeing anything in the Serial Moniotor, that means 

1.UDP packets sent by Alexa are being blocked by a firewall 

2. UDP multicasts are blocked from the router

3. or you are connected to the wrong Wifi.

Try this.

Start a APN on your mobile.
Reconfigure Alexa to connect to APN
Connect the Swicth to APN

Try discovering

solrac21, 2016-12-23:
I tried the APN route and have the same result. i can see these repeating:
Received packet of size 313
From 192.168.43.113, port 48353

Received packet of size 322
From 192.168.43.113, port 48353

over and over but i never can discover the sonoffs.  Both were connected to the APN and show a successful connection.  I will keep trying different things...seriously considering getting the wemos tho....

AndrewsHappy, 2016-12-23:
I know how you feel, that was exactly how I thought once I had programmed it and wasn't getting anywhere. I guess if you can get the Wemos it will prove your system works... Just one small thought, I had an Meru wifi system, which I have just replaced. It didn't like apple iPhones and some of the internet of things I was putting on it. (well known problem apparently)

So just to go back to basics.

You can see on the serial monitor your Sonoff logging onto the network and gives the same IP as what Alexa is on, but when you ask Alexa to find your device and press the button on the sonoff, you do not see any requests on the Serial monitor. in which case it might not be the sonoff that is the problem.

You could try and get hold of aLinkNode R4 and give that a go, slightyly different code, but it is on here somewhere....

Best of Luck.... you are me, this time last week!

solrac21, 2016-12-24:
Thank AndrewsHappy...well i have a list of blocked ports for comcast/xfinity and 1900 was on there.  I have tried various ports and even placing the Sonoff's IP address in the DMZ on the router to no avail.  I tested a diffrent code on the sonoffs to make sure they werent faulty and they worked as they should.

idreamsi, 2017-09-14:
Hi,
Is this script compatible with none physical echo device for example "echosim.io" website or other sites and applications? Who has tested?

Ameengray, 2017-09-26:
I have been trying to get alexa to discover devices and she just wont, my serial monitor will just stop at 

Connecting to WiFi
Connecting .....
Connected to Am
IP address: 192.168.0.108
Begin multicast ..
Udp multicast server started at 239.255.255.250:1900
WebServer started on port: 
80
WebServer started on port: 
81
Adding switches upnp broadcast responder
Adding switch : office lights index : 0
Adding switch : kitchen lights index : 1

and when I ask alexa to discover devices, I don't see anything in the serial monitor. I am on the same network as alexa. I am using the nodemcu cp2102 dev board. I also tried slightly different code with esp8266, I was able to turn on and off the led from the browser but alexa is just not discovering devices. Can please get some help ?

kakopappa, 2017-09-27:
Check whether IGMP / multicast is disabled in your router. 

Also try starting an APN on your mobile and connecting Alexa and dev-board to the same APN.

mortoney, 2017-12-29:
Hi, 
I am trying to create alexa controlled window blinds. I have servo motor and wemos d1-mini. 

I uploaded code to wemos d1-mini and it seems to connected fine to mine wifi. (at this point servo motor is not plugged in to wemos, I just want to make sure alexa have access to it).

Message that I get in console:

Connecting to WiFi
Connecting ........
Connected to VM140913-2G
IP address: 192.168.0.13
Ask Alexa to discover devices

Connecting to UDP
Connection successful
HTTP Server started ..

Now, When Alexa tries to discover devices, she says that non found.

I can see in console that alexa hits wemos server:

192.168.0.10, port 50000
192.168.0.10, port 50000
192.168.0.10, port 50000
192.168.0.10, port 50000
192.168.0.10, port 50000
192.168.0.10, port 50000
192.168.0.10, port 50000
192.168.0.10, port 50000
192.168.0.10, port 50000
192.168.0.10, port 50000
192.168.0.4, port 49170
192.168.0.4, port 49170
192.168.0.4, port 49170
192.168.0.4, port 49170

Any suggestions? is it my router settings?

Thanks

samazw, 2017-12-29:
Do not ask Alexa - use the Alexa App to find your device. This should work.

Best regards
Sascha

mortoney, 2017-12-29:
@samazw Thank for your reply, but I tried both, asking alexa and using the app. In both cases no devices found. But I can see in the console that alexa hits my device with some requests, but for some reason, it does not recognise as a suitable device.

samazw, 2017-12-29:
Hi

Could you please send your coding for analyzing?

Does the initial coding work (Version with issue 22 fixed)?

Best regards
Sascha

mortoney, 2017-12-29:
Code I use is latest code from here: https://github.com/kakopappa/arduino-esp8266-alexa-wemo-switch/blob/master/sinric.ino

Only thing that I changed is wifi connection details, And it seems to connect fine to my wifi.

samazw, 2017-12-29:
Ah ok, please do the following change. This is 100% working with my Setup and even others (copied from the bottomside here:  https://github.com/kakopappa/arduino-esp8266-alexa-multiple-wemo-switch/issues/22   )

My device wasn't discovered because of the code:

if(request.indexOf('M-SEARCH') > 0) {
....
  respondToSearch();
}

Because 'M-SEARCH' is at index 0 (in my case) the above if statement is never true and respondToSearch() is never invoked. I've changed the code to:

if(request.indexOf('M-SEARCH') >= 0) {
....
  respondToSearch();
}

and my Echo has been discovered successfully.

This could be the fix for those with devices that are not discovered.

Best regards
Sascha

mortoney, 2017-12-29:
Thanks. That seems to improve things. I can see that response is being sent.

Sending response to 192.168.0.10
Port : 50000
Response sent !
192.168.0.10, port 50000
Responding to search request ...

Sending response to 192.168.0.10
Port : 50000
Response sent !
192.168.0.10, port 50000
Responding to search request ...

Sending response to 192.168.0.10
Port : 50000
Response sent !

But still alex app and alexa itself say that no devices found. Do I need any skills enable? or any setup todo on the app?

samazw, 2017-12-29:
Hmm,

Do you have at least one „real“ hue device connected to Alexa hub? Furthermore: start search using the App as soon as ESP just connected to your wifi.

I am in holidays on my iPhone so I can do no deeper analyzing (a bit strange on a phone without a computer)... sorry!

Best regards
Sascha

mortoney, 2017-12-29:
I tried restarting wemos d1-mini and searching for devices as soon server started, but no success. 
I don't have any smart home devices connected to my echo yet. But when I ask amazon to search for devices it says to me that I would make sure to enable skill. But I can not enable belkin wemos skill without real switch device.

samazw, 2017-12-29:
There is no further skill required but at least one real Hue device, such as a bulb.

Best regards
Sascha

samazw, 2017-12-29:
One last idea without a real Hue device: use this version (considering also Code change I sent you previously): https://github.com/kakopappa/arduino-esp8266-alexa-multiple-wemo-switch?files=1

Regards
Sascha

mortoney, 2017-12-29:
Oh, ok. Did not see anywhere in readme mentioning this. 

Thanks for your help Sascha

samazw, 2017-12-29:
Good luck!

Best regards
Sascha

jerrishjacob, 2017-12-30:
hi Mortoney,
I am also facing the exact same issue with my echo version 2 and esp8266 nodemcu board. Have you been able to solve this.

kakopappa, 2017-12-30:
Have you tried the latest code? I have made some improvements related not
being able to find

mortoney, 2017-12-30:
@jerrishjacob I still have this issue where alexa cannot discover my wemos d1-mini device. I also using 2nd generation Echo. Maybe that is the reason, it could expect a different response or something.

@kakopappa Yes, I have tried latest version with >= which improved things, Code now sends back a response to Alexa, but Alexa still does not recognise. 

I have not tried a version of the code for multiple devices, I will try to do that later today.

mortoney, 2017-12-30:
Just tested with multi device version of the code and getting same issue:

Got UDP Belkin Request..

Sending response to 192.168.0.1
Port : 29352
Response sent !

Sending response to 192.168.0.1
Port : 29352
Response sent !
Got UDP Belkin Request..

Sending response to 192.168.0.1
Port : 29352
Response sent !

But Alexa does not recognize device.

Is it because I don't have any legit smart device connected, as Sascha suggested HUE light?

kakopappa, 2017-12-30:
IP address : 192.168.0.1

is this your router ip address or alexa ?

mortoney, 2017-12-30:
192.168.0.1 is router IP address

mortoney, 2017-12-30:
Amazon ehco ip: 192.168.0.10

Interesting that single device code sent repond to ip: 192.168.0.10 and multi device code sent repond to ip: 192.168.0.1

in both way alexa can't find device

jerrishjacob, 2017-12-30:
@kakopappa I tried with the updated code.
I am getting the following in the serial monitor: 192.168.1.107 is the IP of alexa.

Connecting to WiFi
Connecting ........
Connected to Jerrish
IP address: 192.168.1.133
Ask Alexa to discover devices

Connecting to UDP
Connection successful
HTTP Server started ..
192.168.1.107, port 50000
Responding to search request ...

Sending response to 192.168.1.107
Port : 50000
Response sent !
192.168.1.107, port 50000
Responding to search request ...

Sending response to 192.168.1.107
Port : 50000
Response sent !
192.168.1.107, port 50000
Responding to search request ...

Sending response to 192.168.1.107
Port : 50000
Response sent !
192.168.1.107, port 50000
Responding to search request ...

Sending response to 192.168.1.107
Port : 50000
Response sent !
192.168.1.107, port 50000
Responding to search request ...

Sending response to 192.168.1.107
Port : 50000
Response sent !
192.168.1.107, port 50000
Responding to search request ...

Sending response to 192.168.1.107
Port : 50000
Response sent !
192.168.1.107, port 50000
Responding to search request ...

Sending response to 192.168.1.107
Port : 50000
Response sent !
192.168.1.107, port 50000
Responding to search request ...

Sending response to 192.168.1.107
Port : 50000
Response sent !
192.168.1.107, port 50000
Responding to search request ...

Sending response to 192.168.1.107
Port : 50000
Response sent !
192.168.1.107, port 50000
Responding to search request ...

Sending response to 192.168.1.107
Port : 50000
Response sent !
192.168.1.1, port 1900
192.168.1.1, port 1900
192.168.1.1, port 1900
192.168.1.1, port 1900
192.168.1.1, port 1900
192.168.1.1, port 1900
192.168.1.1, port 1900
192.168.1.1, port 1900
192.168.1.1, port 1900
192.168.1.1, port 1900
192.168.1.1, port 1900
192.168.1.1, port 1900

[23 more replies]