kakopappa/arduino-esp8266-alexa-wemo-switch #9: Toggle relay request

xbmcnut, 2016-11-16:
Hi,
Is it possible to tweak the code to allow the wemos relay shield to toggle? Want to use it for a garage door controller.

kakopappa, 2016-11-25:
Hi, 

sorry about the late response. 

Right now it is not possible to get the status of the WeMos Relay (https://www.wemos.cc/product/relay-shield.html). I think only option is to define a variable and hold the open/close status then again if you open it manually state become invalid.

xbmcnut, 2016-11-27:
Thanks, appreciate the feedback. But if the status is unknown, you're right, the value is not good.

kakopappa, 2016-11-28:
I think you can use a reed switch to check the status of the door
http://www.ebay.com/bhp/magnetic-reed-switch