kakopappa/arduino-esp8266-alexa-wemo-switch #12: Home Assistant Integration

Banana5kin, 2017-01-03:
Hi,

Great job by the way, got this working with the minimum of fuss with the device added to the Amazon Echo and switching the relay on demand.

Is there any way to control the state of the pins via Home Assistant - Web?  I'd ideally like to be able to control a relay by the Echo and Home Assistant.

Thanks

aneisch, 2017-02-02:
I have made https://github.com/aneisch/Sonoff-Wemo-Home-Assistant/blob/master/sonoff_wemo_homeassistant.ino

This code is compatible with Home Assistant autodiscovery and allows control of Sonoff through emulated hue.

bjhutchins, 2018-09-22:
@Banana5kin 
Did you manage to integrate Alexa into the code at all?
im having trouble

brutus6506, 2023-11-12:
Hi everyone, sorry but I have a question. I made and modified the .INO code and got 10 SW that Alexa identifies perfectly. However, I have a problem with the HA integration. I have installed the WEMO integration and the device is recognized, but it contains a single entity.
What am I doing wrong, which file should I modify?