kakopappa/arduino-esp8266-alexa-wemo-switch #13: Does anyone have this code working with a WiFi configuration manager? thanks

captainchaosrules, 2017-01-11:


Enterius, 2017-03-10:
I tried but I failed. So I have to wait for some help from someone more skilled than me. i am just the beginner :)

hkauroberoi, 2018-01-01:
Yes... I did it and work fine. WiFiManager + Alexa + MQTT

abhinavjha86, 2018-06-02:
Hi,

I integrated WifiManger code with Kaopappa's Wemo emulation code and it works great!!
Here's the sketch...
[sinric_WiFiManager.ino.zip](https://github.com/kakopappa/arduino-esp8266-alexa-wemo-switch/files/2065075/sinric_WiFiManager.ino.zip)

JCMireles, 2019-11-03:
So I have put the code in and I keep getting fail to mount FS on the serial monitor I am new to all this so still learning yes I am a Newbee do I have to change anything cuz all I did was copy and past the code