kakopappa/arduino-esp8266-alexa-wemo-switch #15: Sonoff MQTT Alexa and Siri?

jat80, 2017-02-17:
Hi there,

This is amazing and i'm looking at trying this out on a spare sonoff.

I currently have a few sonoffs and a slampher which i've flashed with he below code:

https://github.com/SuperHouse/BasicOTARelay/blob/master/BasicOTARelay.ino

It uses MQTT to turn it on and off and have that linked to an accessory file linked to HAP-NodeJS to use it with apples homekit software.  Works great.

The one thing with homekit devices and siri though is you have to use your phone to control it.  I would love to use an echo dot in the rooms to also control my hacked sonoff homekit devices and was thinking is there anyway to combine the two?

You code and the above code I'm currently using somehow together?

I can't code so not sure how one would go about this but incorporating MQTT into this code would be absolutely amazing.  A device that both siri and alexa can control!  Maybe finally Alexa and Siri can be friends?  lol

JT.

1rvy, 2017-02-18:
I use Alexa and Homebridge to control several ESP8266 devices around the house, no need for MQTT.

jat80, 2017-02-18:
Good for you but I don't use homebridge.

1rvy, 2017-02-18:
Fair enough, stay struggling with MQTT then!

dougstrickland, 2017-02-18:
I run ESP8266 directly with Alexa, very simple.

jat80, 2017-02-18:
Amazing....

Read the OP.

1rvy, 2017-02-18:
Good luck getting help with coding your sketch with that attitude.   We are informing you that we have acheived your end result through different means, but if you don't want to know how then you best start learning how to do it on your own.

jat80, 2017-02-18:
Lol. What a douche'.

jat80, 2017-03-05:
Found these tutorials on how to control the sonoff via siri and alexa.  Enjoy :)

https://www.youtube.com/watch?v=n4MDRm2yAJg&t

https://www.youtube.com/watch?v=M4qHXoAyrRs&t

closed 2017-03-05 by jat80