kakopappa/arduino-esp8266-alexa-wemo-switch #18: Add a pushbutton switch to toggle relay - NOT Sonos

dougstrickland, 2017-03-01:
I am trying to add a pushbutton switch on one of the GPIO pins, then toggle the relay between on-off states.  This makes it possible to Manually turn the relay on or off with the push of a button, and not having to use Alexa every time.

I am saving the current on/off state, so simply flip it whenever the button is pushed, but I can't get the switch to be read.

Any suggestions are appreciated!

kakopappa, 2017-03-02:
Push buttons often generate spurious open close singles when pressed. it's called Debounce. You have to handle it properly to get the push button working. 

https://www.youtube.com/watch?v=jYOYgU2vlSE

I like to recommend you to turn on /off a LED light with a push button first and then change the code to turn on /off relay.

https://learn.sparkfun.com/tutorials/sik-experiment-guide-for-arduino---v32/experiment-5-push-buttons

dougstrickland, 2017-03-02:
Thanks for the suggestion. I've debounced the switch, and checked it with a scope. The switch is working correctly. I don't think it's getting to my code, seems to be looping through something else.  Would you recommend putting it at the top of the loop script, or somewhere in the middle.  I currently have it at the bottom.

Svengali86, 2017-03-02:
Hi I did this works well.

`#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <WiFiUdp.h>
#include <functional>

void prepareIds();
boolean connectWifi();
boolean connectUDP();
void startHttpServer();
void turnOnRelay();
void turnOffRelay();

const char* ssid = "******";
const char* password = "*****";

unsigned int localPort = 1900;      // local port to listen on

WiFiUDP UDP;
boolean udpConnected = false;
IPAddress ipMulti(239, 255, 255, 250);
unsigned int portMulti = 1900;      // local port to listen on

ESP8266WebServer HTTP(80);
 
boolean wifiConnected = false;

char packetBuffer[UDP_TX_PACKET_MAX_SIZE]; //buffer to hold incoming packet,

String serial;
String persistent_uuid;
String device_name;

const int relayPin = 2;
const int switchPin = 3;

boolean cannotConnectToWifi = false;

int switchState = 0;

void setup() {
  Serial.begin(115200);

  // Setup Relay
  pinMode(relayPin, OUTPUT);

  // Setup Switch
  pinMode(switchPin, INPUT);  

  turnOffRelay();
  switchState = 0;
  Serial.println("Setting relay to off - setting value to 0");
  
  prepareIds();
  
  // Initialise wifi connection
  wifiConnected = connectWifi();

  // only proceed if wifi connection successful
  if(wifiConnected){
    udpConnected = connectUDP();
    
    if (udpConnected){
      // initialise pins if needed 
      startHttpServer();
    }
  }  
}

[... 262 lines cut ...]
  boolean state = false;
  
  Serial.println("");
  Serial.println("Connecting to UDP");
  
  if(UDP.beginMulticast(WiFi.localIP(), ipMulti, portMulti)) {
    Serial.println("Connection successful");
    state = true;
  }
  else{
    Serial.println("Connection failed");
  }
  
  return state;
}

void turnOnRelay() {
 digitalWrite(relayPin, HIGH); // turn on relay with voltage HIGH 
 switchState = 1;
}

void turnOffRelay() {
  digitalWrite(relayPin, LOW);  // turn off relay with voltage LOW
  switchState = 0;
}`

dougstrickland, 2017-03-04:
I'm getting a bunch of compile errors, starting with an empty #include on line 4.  I've tried working through the syntax errors, but I seem to be making it worse.  Can you try compiling my script and see what you get?

#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <WiFiUdp.h>
// #include

void prepareIds();
boolean connectWifi();
boolean connectUDP();
void startHttpServer();
void turnOnRelay();
void turnOffRelay();

const char* ssid = "buddha";
const char* password = "0305196001161955";

unsigned int localPort = 1900; // local port to listen on

WiFiUDP UDP;
boolean udpConnected = false;
IPAddress ipMulti(239, 255, 255, 250);
unsigned int portMulti = 1900; // local port to listen on

ESP8266WebServer HTTP(80);

boolean wifiConnected = false;

char packetBuffer[UDP_TX_PACKET_MAX_SIZE]; //buffer to hold incoming packet,

String serial;
String persistent_uuid;
String device_name;

const int relayPin = 2;
const int switchPin = 3;

boolean cannotConnectToWifi = false;

int switchState = 0;

void setup() {
Serial.begin(115200);

// Setup Relay
pinMode(relayPin, OUTPUT);

// Setup Switch
pinMode(switchPin, INPUT);

turnOffRelay();
switchState = 0;
Serial.println("Setting relay to off - setting value to 0");

prepareIds();

// Initialise wifi connection
wifiConnected = connectWifi();

// only proceed if wifi connection successful
if(wifiConnected){
udpConnected = connectUDP();

if (udpConnected){
  // initialise pins if needed 
  startHttpServer();
}
}
}

[... 260 lines cut ...]
boolean state = false;

Serial.println("");
Serial.println("Connecting to UDP");

if(UDP.beginMulticast(WiFi.localIP(), ipMulti, portMulti)) {
Serial.println("Connection successful");
state = true;
}
else{
Serial.println("Connection failed");
}

return state;
}

void turnOnRelay() {
digitalWrite(relayPin, HIGH); // turn on relay with voltage HIGH
switchState = 1;
}

void turnOffRelay() {
digitalWrite(relayPin, LOW); // turn off relay with voltage LOW
switchState = 0;
}

Svengali86, 2017-03-04:
Hi Doug sent you my code on email, I think pasting it in here messes it up somehow 

cheers

closed 2017-03-05 by dougstrickland