kakopappa/arduino-esp8266-alexa-wemo-switch #23: Aruna , need some help.

dkelly324, 2017-10-10:
I'm trying open and close my vertical blinds with your wemo emulator.  Everything works fine until I increase the run time on either a servo or stepper.  Alexa will tell me something went wrong and do another run.  I've added a 6 second delay after test lights on and off in the single switch example and the same thing happens.  Help

kakopappa, 2017-10-10:
can you share your code? You must be blocking the loop()

dkelly324, 2017-10-10:
Ok, I’ll get it to you tomorrow.

Dennis Kelly
dkelly324@gmail.com

dkelly324, 2017-10-10:
Aruna, just remember I'm no C++ guy so I've pieced this together the best I
could.  The sketch labeled blinds contains your original emulator but they
both do the same thing.  Once the run time on the stepper or the servo is
increased to about 3 revolutions, it starts having problems.

Thank you so very much for your help.
Dennis Kelly    dkelly324@gmail.com

Stepper code-
#include <ESP8266WiFi.h>
#include "Stepper.h"
#include "WemoSwitch.h"
#include "WemoManager.h"
#include "CallbackFunction.h"
int Pin0 = D1;//definition digital 11 pins as pin to control the IN1
(ULN24L01)
int Pin1 = D2;//definition digital 10 pins as pin to control the IN2
(ULN24L01)
int Pin2 = D3;//definition digital 9 pins as pin to control the IN3
(ULN24L01)
int Pin3 = D4;
int _step = 200;
int _speed = 1;
// prototypes
boolean connectWifi();
//on/off callbacks
void lightOn();
void lightOff();
//------- Replace the following! ------
char ssid[] = "BelDen";       // your network SSID (name)
char password[] = "xxx*$#";  // your network key
WemoManager wemoManager;
WemoSwitch *light = NULL;
const int ledPin = BUILTIN_LED;
void setup()
{
  Serial.begin(115200);
  // Set WiFi to station mode and disconnect from an AP if it was Previously
  // connected
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  delay(100);
  // Attempt to connect to Wifi network:
  Serial.print("Connecting Wifi: ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(500);
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  IPAddress ip = WiFi.localIP();
  Serial.println(ip);
  wemoManager.begin();
  // Format: Alexa invocation name, local port no, on callback, off callback
  light = new WemoSwitch("test", 80, lightOn, lightOff);
  wemoManager.addDevice(*light);
  pinMode(ledPin, OUTPUT); // initialize digital ledPin as an output.
  pinMode(Pin0, OUTPUT);//Set digital 8 port mode, the OUTPUT for the output
  pinMode(Pin1, OUTPUT);//Set digital 9 port mode, the OUTPUT for the output
  pinMode(Pin2, OUTPUT);//Set digital 10 port mode, the OUTPUT for the
output
  pinMode(Pin3, OUTPUT);//Set digital 11 port mode, the OUTPUT for the
output
  delay(10);
  digitalWrite(ledPin, HIGH); // Wemos BUILTIN_LED is active Low, so high
is off
}
[... 375 lines cut ...]
  "EXT:\r\n"
  "LOCATION: http://" + String(s) + ":80/setup.xml\r\n"
  "OPT: \"http://schemas.upnp.org/upnp/1/0/\"; ns=01\r\n"
  "01-NLS: b9200ebb-736d-4b93-bf03-835149d13983\r\n"
  "SERVER: Unspecified, UPnP/1.0, Unspecified\r\n"
  "ST: urn:Belkin:device:**\r\n"
  "USN: uuid:" + persistent_uuid + "::urn:Belkin:device:**\r\n"
  "X-User-Agent: redsonic\r\n\r\n";

  UDP.beginPacket(UDP.remoteIP(), UDP.remotePort());
  UDP.write(response.c_str());
  UDP.endPacket();
  if (debug) {
    Serial.println("Response sent !");
  }
}

void configModeCallback(WiFiManager *myWiFiManager) {
  Serial.println("Entered config mode");
  Serial.println("Soft AP's IP Address:");
  Serial.println(WiFi.softAPIP());
  Serial.println("WiFi Manager: Please connect to AP:");
  Serial.println(myWiFiManager->getConfigPortalSSID());
  Serial.println("To setup WiFi Configuration");
}

kakopappa, 2017-10-11:
your problem is you are using delay(). Delay will block the main loop and stops everything (except interrupts)  try using a timer.

https://playground.arduino.cc/Code/AvoidDelay
https://forum.arduino.cc/index.php?topic=355651.0