kakopappa/arduino-esp8266-alexa-wemo-switch #25: Is there a way to make this work with Google Home?

atomi, 2018-01-01:
Can someone point me to a Google home version? Or some documentation on the subject? 

Or consider a feature request for this I guess.

samazw, 2018-01-01:
Google is different - not a feature request but a new project. Interested in programming yourself, of just costless consuming?

Best regards
Sascha

atomi, 2018-01-01:
I was just curious. Do you have some links on similar Google Home projects for esp8266?

samazw, 2018-01-01:
Well, just use Google. 1st entry http://tinkerman.cat/using-google-assistant-control-your-esp8266-devices/

But I seriously think I‘ll not use Google to get results for you. Or - again -  be a programmer and share 😉

 Best regards
Sascha

atomi, 2018-01-02:
Haha, I thought you were more familiar with Google Home hardware which is why I asked. No worries. 

I think an easy way to use esp with google would be an ifttt applet that calls a webhook hosted on the esp. I'll play around with it on my side. 

Thanks.

closed 2018-01-02 by atomi

samazw, 2018-01-02:
Do it ;-)

Best regards
Sascha