kakopappa/arduino-esp8266-alexa-wemo-switch #26: Echo Plus (2. Gen) discovery Problem

JanB97, 2018-01-03:
I had problems discovering my WeMos D1 Mini with Alexa.
After reading a bit through some Fauxmo Issues I found something that was different and seems to work there.
I had to change from this:
String response = 
         "HTTP/1.1 200 OK\r\n"
         "CACHE-CONTROL: max-age=86400\r\n"
         "DATE: Fri, 15 Apr 2016 04:56:29 GMT\r\n"
         "EXT:\r\n"
         "LOCATION: http://" + String(s) + ":80/setup.xml\r\n"
         "OPT: \"http://schemas.upnp.org/upnp/1/0/\"; ns=01\r\n"
         "01-NLS: b9200ebb-736d-4b93-bf03-835149d13983\r\n"
         "SERVER: Unspecified, UPnP/1.0, Unspecified\r\n"
         "ST: urn:Belkin:device:**\r\n"
         "USN: uuid:" + persistent_uuid + "::urn:Belkin:device:**\r\n"
         "X-User-Agent: redsonic\r\n\r\n";

to this:
String response = 
         "HTTP/1.1 200 OK\r\n"
         "CACHE-CONTROL: max-age=86400\r\n"
         "DATE: Fri, 15 Apr 2016 04:56:29 GMT\r\n"
         "EXT:\r\n"
         "LOCATION: http://" + String(s) + ":80/setup.xml\r\n"
         "OPT: \"http://schemas.upnp.org/upnp/1/0/\"; ns=01\r\n"
         "01-NLS: b9200ebb-736d-4b93-bf03-835149d13983\r\n"
         "SERVER: Unspecified, UPnP/1.0, Unspecified\r\n"
         "ST: ssdp:all\r\n"
         "USN: uuid:" + persistent_uuid + "::upnp:rootdevice\r\n"
         "X-User-Agent: redsonic\r\n\r\n";

My device is discovering fine now, maybe this helps someone.

Btw thank you for the awesome work and I am sorry if this is the wrong place for this. :)

ATXdesigns, 2018-04-10:
what file is this in?

JanB97, 2023-01-03:
Closing this because of sinric pro.

closed 2023-01-03 by JanB97