kakopappa/arduino-esp8266-alexa-wemo-switch #27: not discovering

jxer, 2018-01-04:
im using

https://www.ebay.com/itm/NodeMCU-V3-Lua-WIFI-Module-ESP8266-32M-Extra-Memory-Flash-USB-Serial-CH340G/222612803340?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

and im getting no where. everything seems to be working just no discovery by alexa

anywhere i can troubleshoot this?

i tried my normal wifi and a hotspot 

all i get is

Connecting to WiFi
Connecting ....
Connected to VWifi
IP address: 10.26.0.10
Ask Alexa to discover devices

Connecting to UDP
Connection successful
HTTP Server started ..

jxer, 2018-01-04:
some more info, very confused, if i browse to the ip/setup.xml it comes right up

`<?xml version="1.0"?>

-<root>

-<device>

<deviceType>urn:Belkin:device:controllee:1</deviceType>

<friendlyName>tv</friendlyName>

<manufacturer>Belkin International Inc.</manufacturer>

<modelName>Socket</modelName>

<modelNumber>3.1415</modelNumber>

<modelDescription>Belkin Plugin Socket 1.0</modelDescription>

<UDN>uuid:Socket-1_0-38323636-4558-4dda-9188-cda0e691c5a8</UDN>

<serialNumber>221517K0101769</serialNumber>

<binaryState>0</binaryState>

-<serviceList>

-<service>

<serviceType>urn:Belkin:service:basicevent:1</serviceType>

<serviceId>urn:Belkin:serviceId:basicevent1</serviceId>

<controlURL>/upnp/control/basicevent1</controlURL>

<eventSubURL>/upnp/event/basicevent1</eventSubURL>

<SCPDURL>/eventservice.xml</SCPDURL>

</service>

</serviceList>

</device>

</root>`

jxer, 2018-01-04:
let me add something, i do not currently have any form of dot/echo etc. is that required before any device will show up? im not getting any error so i have no idea

stnkl, 2018-01-04:
I'm experiencing the same problem. Don't own a dot yet either.

jxer, 2018-01-04:
I'm being told that a dot/echo is required for the as the discovery packet comes from the dot/echo, not the app. i wish someone could confirm

kakopappa, 2018-01-08:
Use sinric.com if you have discovery issues. This is one of the reasons i built this. Instructions on the home page