kakopappa/arduino-esp8266-alexa-wemo-switch #28: Pin help

civicbynature, 2018-01-24:
I must be missing something super simple. 

                        Format: Alexa invocation name, local port no, on callback, off callback
on line " office = new Switch("office lights", 80, officeLightsOn, officeLightsOff);"

What pin is "80" ?

I can seem to figure out which pin is toggling high/low. 

I have tried replacing "80" with "1", "D1" etc but can not seem to figure out how to set which pin this line in toggling. 

its completely recognized by the network and alexa is saying it's turning it on and off but i can't figure out which pin is actually toggling. 

Please and thank you for your great work here.

samazw, 2018-01-24:
Hi

80 is the port -> https://en.m.wikipedia.org/wiki/Port_(computer_networking)

All best...

MadMaxDallas, 2018-04-03:
At top of the file is the definition.
const int relayPin = 5;  // D1 pin. More info: https://github.com/esp8266/Arduino/blob/master/variants/d1_mini/pins_arduino.h#L49-L61
Regards