kakopappa/arduino-esp8266-alexa-wemo-switch #31: Instructions to Alexa result in the reverse

BLC-Bob, 2018-02-18:
Thanks, Aruna, for the sketch - it's the only one I have found that works. I am new to all this. I have a 5v 2-Channel Relay Module (but I'm just using one relay at the moment) triggered by your sketch running on an ESP-1s.  Problem is that when I tell Alexa to turn the relay on, the relay turns off and vice versa. I think this is because this relay model is triggered by the ESP-01s output pin (I'm using GPIO02) being set LOW rather than HIGH (the opposite to the instructions in the sketch), by sinking the voltage from VCC to the IN1 pin.  So I guess to correct this I need change the instructions round in the sketch. Most of the instructions in this sketch are way beyond my current level of understanding . . . I'm sure I'll get there one day! in the meantime, which lines would I need to change to get the relay to turn on when I say "ON" and off when I say "OFF", please. ;-{)

slavino, 2018-02-28:
Make sure when the relay is on - if the pin needs to be OUTPUT high or low.
 
Depending on board you might use NPN transistor (or optoisolated component or logic level shifter or...) to steer 5V logic from 3V3 board.

tenthirtynine, 2018-11-09:
> Make sure when the relay is on - if the pin needs to be OUTPUT high or low.
> 
> Depending on board you might use NPN transistor (or optoisolated component or logic level shifter or...) to steer 5V logic from 3V3 board.

Hey, I have this issue using the D1 Mini, what NPN transistor would you use? Pretty sure its a 3V board

dougstrickland, 2018-11-09:
2N3904