kakopappa/arduino-esp8266-alexa-wemo-switch #33: Alexa app can't discover my Alexa Dot (2nd Gen) device

GaryLittlemore, 2018-04-04:
I've uploaded the .ino file to the Wemos board. I can see it connected to my WiFi, and I can connect to the IP address it's connected too but my Alexa app isn't finding the device. Below is my code, does anything look wrong?

https://pastebin.com/d2BQVjYu﻿

kakopappa, 2018-04-05:
This project does not support Alexa app.  If you want to use app, try my other project https://sinric.com

GaryLittlemore, 2018-04-05:
Thanks, watching this James Bruton youtube video he seems to think it does or am I missing something?

https://youtu.be/wMpWQ4znTqQ

sabhay1996, 2018-12-15:
> This project does not support Alexa app. If you want to use app, try my other project https://sinric.com

Project is working for me, but problem is every time nodemcu restarts, alexa is not able to recognise the device giving the error, unable to respond. After pushing the RST Button, device is back and recognisable.

samazw, 2018-12-15:
Do you habe GPIO-9 in use? This one must not be used. Also consider a pullup resistor on reset. Just 4 + 5 work stable.

Best regards
Sascha

sabhay1996, 2018-12-16:
I am still using 5 as it is pre defined in the project.

samazw, 2018-12-16:
Which Module do you use (esp - sure... which concrete model?) any special circuit?
Best regards
Sascha

sabhay1996, 2018-12-16:
I am using nodemcu lolin v3.
Flash Size: 4M (1M SPIFFS)
IwIP: 2v Higher Bandwidth
Upload Speed: 115200
Board: NodeMCU 1.0
VTable: Flash

samazw, 2018-12-16:
Ah ok. Lolin v3 often comes up with reset issues and I was able to reproduce your issue. So GPIO0 should be pulled up to 3.3V using a 10k resistor AFTER being programmed. Keep in mind that re-programming using USB is only possible when removing this pullup resistor. So a jumper makes sense here. 
Best regards
Sascha

> Am 16.12.2018 um 12:08 schrieb sabhay1996 <notifications@github.com>:
> 
> lolin v3

sabhay1996, 2018-12-16:
Please correct me if did something wrong, I tried using 2 jumper and 10k between them, from gpio 0/D3 to 3.3v. 
I have configured the project with gpio 4/D2 and using alternative 3.3V and GND for the relay.
But still I need to press the reset button.

samazw, 2018-12-16:
Please Check if there is a soldered resistor on your board. If so either remove it or try an 3k3 instead of the 10k... good luck!

sabhay1996, 2018-12-17:
Thanks buddy for promptly replying to help me.
Still no success, after uploading the sketch, it connects to the wifi, when I reboot, it doesn't connect I need to manually press the reset switch. 
I tried 3K3 resistance from GPIO0 to 3.3V
I tried 10K resistance from GPIO0 to 3.3V

GPIO5 to GND as well.
Tried same with another boards as well no success.
Maybe I am doing something please help?