kakopappa/arduino-esp8266-alexa-wemo-switch #39: Stopped working with latest Alexa update

pacostello, 2018-12-05:
Wemos D1 mini and relay setup has been working consistently until today. 
The Alexa app can no longer find it when detecting devices

samazw, 2018-12-05:
Any changes in your WiFi? Could you provide more details, such as WiFi pw changed, type of router / accesspoint, visibility of the Wemos in your LAN, did Alexa an update etc. ? Mine still working fine (multiple, different ESP devices), so Alexa still seems to be stable.

pacostello, 2018-12-05:
No changes to my network so all I can think is an Alexa update. I can see and ping the wemos as normal. 
I am in the UK so may get different updates to other countries

pacostello, 2018-12-05:
Somehow it is now working. Switched it off and on again and it's now detecting it.
Sorry to waste your time.

samazw, 2018-12-05:
No waste of time - no question is stupid but some answers are ;-). Be happy - it works :-)