kakopappa/arduino-esp8266-alexa-wemo-switch #44: arduino esp8266 script changed in Alexa

TexAcoon, 2019-11-04:
kakopappa - 
I have used this script for over a year and worked so well I was planning on adding more devices to my home automation.  Not sure what has gone wrong when asking Alexa to turn on the appliance it "states there is a malfunction" OR "cant find the appliance" (depending on what day it is and how good she feels.  

I have not made any chances to the software and still can not get Alexa do discover any of my WeMos ESP8266s ... tried looking in "Serial Monitor" and do not see anything working.    Has something changed in the Amazon Alexa for the system not to recognize the WeMos ESP8266 any longer?

kakopappa, 2019-11-04:
Hi @TexAcoon  

Emulation seems to have issues with new Alexa firmware updates. I think a smart home skill is the right way to do this. Do you mind giving a https://sinric.pro a try ?

TexAcoon, 2019-11-04:
Sure I will download it and try it out .. Will let you know what the status is when complete.

TexAcoon, 2019-11-04:
You have to pay for that service - I thought the whole idea of home automation using what we have is to only pay for the components and not the monthly programing.  However, there is a free month involved and really interested what is going on with this ..

TexAcoon, 2019-11-04:
Kakopappa - 
Do I use the same Arduino script or do I have to change another script ..  for the smart devices that are built in like the Phillips light bulbs etc ...   I can see how the connection would work ..   Not sure if I have to add a new script to the WeMos or use the one you already created ...   Thanks

kakopappa, 2019-11-04:
Pro provides 5 devices for free for life and additional device is $3 per year!

you can find a sample sketch for switch here
https://github.com/sinricpro/esp8266-esp32-sdk/tree/master/examples/switch

Fill in the functions with the code you had already. shouldn't take longer than 5 mins

TexAcoon, 2019-11-04:
I have signed up and tried loading the software - Before there was not any issue uploading your script to "NodeMCU  1.0(ESP-12E)Module ..   Now it gives me an error message: "exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module)."
I have tried several difference boards and they say the same thing ..    What do you suggest?

kakopappa, 2019-11-04:
Need more details

Are you using Arduino or PlatformIO?

Can you please upload a screenshot ?

TexAcoon, 2019-11-04:
I'm using Arduino never have tried the PlatformIO... 
Here is my error message I am getting at this time..
[https://www.dropbox.com/home?preview=esp8266+errpr.PNG]

kakopappa, 2019-11-04:
I think you forgot to share the image properly.

It shows me this when i open your link

[image]

kakopappa, 2019-11-04:
Maybe you do not have the dependencies installed 

[image]

https://github.com/sinricpro/esp8266-esp32-sdk#dependencies

TexAcoon, 2019-11-04:
Aruna,
Excellent job! identifying my issue! ..   I downloaded all three of the dependencies and did a Arduino code "verify" on the code ..That took care of it! ... Did upload the code just as a double check to the NodeMCU and loaded like it should.

I signed up at Sinric Pro and added a device naming it "Fireplace" ..  

I did receive a code for my account and one for the device "Fireplace" .. but it does not describe exactly what code is for which of the define(s)...  
#define SOCKET_AUTH_TOKEN "??????"#define SIGNING_KEY       "YOUR-SIGNING-KEY"#define SWITCH_ID  "Fireplace"  ????

Thanks Doug (TexAcoon)

 
 

https://github.com/sinricpro/esp8266-esp32-sdk#dependencies

TexAcoon, 2019-11-04:
Here is a copy of the error message incase someone else runs across this same issue ...
    On Monday, November 4, 2019, 12:51:17 PM CST, Doug Crawford <rocknbarc@yahoo.com> wrote:  
 
  Aruna,
Excellent job! identifying my issue! ..   I downloaded all three of the dependencies and did a Arduino code "verify" on the code ..That took care of it! ... Did upload the code just as a double check to the NodeMCU and loaded like it should.

I signed up at Sinric Pro and added a device naming it "Fireplace" ..  

I did receive a code for my account and one for the device "Fireplace" .. but it does not describe exactly what code is for which of the define(s)...  
#define SOCKET_AUTH_TOKEN "??????"#define SIGNING_KEY       "YOUR-SIGNING-KEY"#define SWITCH_ID  "Fireplace"  ????

Thanks Doug (TexAcoon)

 
 

https://github.com/sinricpro/esp8266-esp32-sdk#dependencies

kakopappa, 2019-11-05:
#define SOCKET_AUTH_TOKEN -> In the Left menu -> Goto Credentails tab -> Copy APP KEY
#define SIGNING_KEY    -> In the Left menu -> Goto Credentails tab -> Copy APP SECRET

[image]

"#define SWITCH_ID  --> In the left menu -> Devices -> Device you created has an id like ID: 5d7exx18cxxdabxxca420bxx 

[image]

TexAcoon, 2019-11-05:
Thanks for the info on the different keys and which they represent ..   I will keep a copy of this email for future reference ...
Seems the unit works fine turn on and turn off in Sinric Pro ..   But Alexa can not locate the device when I ask "discover devices" - What do  you suggest?

kakopappa, 2019-11-05:
have you enabled the Sinric Pro skill and linked it? it's in the home page https://sinric.pro/

[image]

After linking the account you should be able to see your device in Alexa app

TexAcoon, 2019-11-05:
Aruna ...  I really do appreciate you helping me on this project ...  Seems a bit confusing dealing with another software... (but its new to me and now feel I have a better handle on it for future devices and I Thank You!)
However, I looked on my cell phone under my devices through the Alexa App ....  Alexa is recognizing the device and turning it on an off on my cell but not on the ESP8266.... Apparently it's not linking back to the ESP8266 for some reason for switching..

kakopappa, 2019-11-06:
np, 

Yesterday, there were some problems with the sevice and it's fixed now. If still does not work

Please enable Debug Info, to do that 

eg:

#define DEBUG_ESP_PORT Serial
#define NODEBUG_WEBSOCKETS
#define NDEBUG

#include "SinricPro.h"
.......

This enables Debug on Serial port. Please post your debug output here

TexAcoon, 2019-11-06:
OK Thanks ...  I will give that a try

TexAcoon, 2019-11-06:
Still cannot connect through Alexa ..  Do not even see the ESP8266 on my Network i.e., internet connections.  ALEXA reconizes "fireplace" and turns it on and off fine ...   Sinric is also working from the Alexa online perfect ...  I can see it on my cell flip off and on ..  But it's not activating the ESP8266 for my switch ???? 
Here is the script upload after I installed the debug info: All seems to look good:

Sketch uses 425722 bytes (40%) of program storage space. Maximum is 1044464 bytes.Global variables use 30452 bytes (37%) of dynamic memory, leaving 51468 bytes for local variables. Maximum is 81920 bytes.esptool.py v2.62.6esptool.py v2.6Serial port COM8Connecting....Chip is ESP8266EXFeatures: WiFiMAC: 2c:3a:e8:43:91:04Uploading stub...Running stub...Stub running...Configuring flash size...Auto-detected Flash size: 4MBCompressed 429872 bytes to 311100...
Writing at 0x00000000... (5 %)Writing at 0x00004000... (10 %)Writing at 0x00008000... (15 %)Writing at 0x0000c000... (21 %)Writing at 0x00010000... (26 %)Writing at 0x00014000... (31 %)Writing at 0x00018000... (36 %)Writing at 0x0001c000... (42 %)Writing at 0x00020000... (47 %)Writing at 0x00024000... (52 %)Writing at 0x00028000... (57 %)Writing at 0x0002c000... (63 %)Writing at 0x00030000... (68 %)Writing at 0x00034000... (73 %)Writing at 0x00038000... (78 %)Writing at 0x0003c000... (84 %)Writing at 0x00040000... (89 %)Writing at 0x00044000... (94 %)Writing at 0x00048000... (100 %)Wrote 429872 bytes (311100 compressed) at 0x00000000 in 27.5 seconds (effective 125.2 kbit/s)...Hash of data verified.
Leaving...Hard resetting via RTS pin...

kakopappa, 2019-11-06:
Can you drop your email here please?

TexAcoon, 2019-11-06:
Seems Sinric Pro is a bit glitchy ...  I have left the page open and see it shows my device going ONLINE then OFFLINE and back and forth ..   I know the software still shows in BETA is there a reason why it's losing connection?

kakopappa, 2019-11-06:
Checking

kakopappa, 2019-11-06:
According to your profile settings, you have not installed the skill.  Please make sure you have linked "Sinric Pro" skill.. not "Sinric" Skill

[image]

TexAcoon, 2019-11-06:
LOL! ... Well that had to be part of the problem ..  I had installed skill Sinric instead of Sinric Pro ..  Yup I can see how this makes a difference ...
However, I asked ALexa to discover devices and still cannot connect to my ESP8266...  However, when I asked Alexa to Turn on or Turn Off the fireplace it stated "it's not responding"  which was different than before.  I would usually get from ALexa "OK"
Im starting to wonder if I have a bad ESP8266..

kakopappa, 2019-11-06:
Yes, Alexa integration seems to be fine now. your problem is there are no devices connected with device id "5dxxxxxx26505c"

Please make sure ESP device id is correct

TexAcoon, 2019-11-06:
ok... I'm at a loss now ..  I thought through the sketch in the ESP8266 it identified the device number then identifies it through Sinric Pro.. for automation with ALexa.  
With my ESP8266 not being recognized none of this works correctly ... ???
At this time I'm not sure why my ESP8266 is not coming up ...  If I remember in the past using the old sketch without Sinric Pro I could see my device on my network.
What should I try to do next?

kakopappa, 2019-11-06:
Can I connect to your pc Teamviewer / quick assist and take a look?

TexAcoon, 2019-11-06:
Yes please do ...

kakopappa, 2019-11-06:
support@sinric.com - please drop me an email when you have a chance.

clox-coder, 2019-11-08:
@kakopappa I set everything up a few weeks ago. I am able to make everything discoverable.  But when i say "Turn On" It says this device does not have the capability.  But i am able to discover etc .  it showing up as a philips hue Bulb. I signed up for Sinric Pro.  But why cant i just use the code instead of the Sinric service?

TexAcoon, 2019-11-08:
Here is what was wrong with mine:
1) Alexa Skills has to have Sinric Pro - NOT - Sinric (that also caused a lot of confusion to Alexa)2) Sinric Pro.com - Personal codes have to be correct and in proper location in your Arduino Script.3) We added a couple of changes to my code that made it work see highlight:
#include <Arduino.h>#ifdef ESP8266 #include <ESP8266WiFi.h>#endif #ifdef ESP32   #include <WiFi.h>#endif
#define DEBUG_ESP_PORT Serial#define NODEBUG_WEBSOCKETS#define NDEBUG
#include "SinricPro.h"#include "SinricProSwitch.h"

#define WIFI_SSID         "Add your WIFI SSID"#define WIFI_PASS         " Add your WIFI PW"#define SOCKET_AUTH_TOKEN "token code"#define SIGNING_KEY       "Sign in key"#define SWITCH_ID         "Your device number you have in Sinric Pro for the named device"
#define BTN_FLASH 0#define SERVER_URL "iot.sinric.com"#define SERVER_PORT 80 
#define HEARTBEAT_INTERVAL 300000 // 5 Minutes 
bool myPowerState = false;unsigned long lastBtnPress = 0;
The rest of the script is the same....... 
_________________________________________________________________________
After battling this for about three days we finally got mine to work .. However, another part of my issue was a weak WIFI system where I was located. Since then that has been upgraded.
Please let us know if these changes worked for you ..  Figured Aruna worked is butt off to help get mine going and I am passing over information that got my system going to others.Overall my system is working like a champ .. and getting ready to add a couple more devices.
Doug (TexAcoon)

clox-coder, 2019-11-08:
@TexAcoon I believe i can get that working.  I was just curious of anther way around it. Anyway to setup a local server on network etc. So i dont have to run anything through an API.  I like to have some control. LOL

kakopappa, 2019-11-08:
@clox-coder Hi, 

With Alexa skill,  there is no emulation. Emulation is not the right way to do this (my opinition) and Amazon Skill is. Emulation tends to break when Alexa updates firmware and sometimes devices go missing. It used to work but lately it is not.

1. Disable **Sirnic** Skill if you have used it before.
2. Create a Sinric Pro account.
2. Link **Sinric Pro** skill. 
3. Login to portal from https://sinric.pro and create a new device.
4. Download the Sketch 
https://github.com/sinricpro/esp8266-esp32-sdk/blob/master/examples/Switch/Switch.ino

5. Install Sinric Pro library from Arduino Library manager 

[image]

6. Copy the APP KEY, APP SECRET from Credentails tab on your left and update example sketch

7. Copy the Device Id from Devices tab and update the Sketch and Flash. 

When the device connect to the server, you will see a notification appear.

clox-coder, 2019-11-08:
@kakopappa i appreciate it.  I understand how to configure the setup.  I was just looking for a local way to do this.  Creating a local server etc.

kakopappa, 2019-11-08:
ok

kakopappa, 2019-11-08:
We do have plans to support offline with Sinric Pro Android app. Right now we are focusing on diagnosing and improving the online version.

TexAcoon, 2019-11-08:
Aruna,
Just a note when you go to "ROOMS" in Sinric Pro and want to edit the one of the rooms - it does not open up and allow you to edit.
Doug

kakopappa, 2019-11-08:
Thanks for letting me know. I will fix this in the next update

TexAcoon, 2019-11-08:
Just a thought on next update as well: 
When you go to "NEW DEVICE" and select "DEVICE TYPE" ...  There are several Sinric Built in devices selections.  I am working on my garage door ALEXA opening and there is not one listed.
Also on a garage door you would OPEN versus turn ON same with CLOSE versus turn OFF.   
Do you change the script "on" for "open" and "off" for "close" to make it work? (Sounds simple to me - but didn't work)   Because the switch ID in Sinric Pro only states On and Off
Doug

kakopappa, 2019-11-08:
Alexa doesn’t support open / close commands as of now. Only workaround is
to create a routine to do this as far as I know

sivar2311, 2019-11-08:
What about using smartlock for this?

TexAcoon, 2019-11-08:
Smart Lock · SinricPro

| 
| 
|  | 
Smart Lock · SinricPro

 |

 |

 |

 

13:05:

sivar2311, 2019-11-08:
Sorry, i read and answered this via email...that's why the post was that long

sivar2311, 2019-11-10:
I made a simple example using a SinricPro SmartLock as GarageDoor.
Have a look at my [YouTube Video](https://www.youtube.com/watch?v=xhg61pQe67A).

Please use #include "SinricProLock.h" to include SmartLock device.

TexAcoon, 2019-11-10:
Boris ...
I really like that! .. and with a code ...   Would like a copy of the script if it's available.Hope one day Alexa understands OPEN and CLOSE since they are common words used each day for various task.
ThanksDoug

sivar2311, 2019-11-10:
Hi Doug!

Here i prepared the sketch for you. It's simple as Switch example...except it's using SinricProLock instead of SinricProSwitch.

Voice pin is required by alexa if you want to use voice activated unlock. Add a SmartLock on portal.sinric.pro. After it is discovered, go to alexa app and configure your smart lock device. You have  to explicit enable voice unlock and you have to define a pin!

```C++
/*
 * If you encounter any issues:
 * - enable serial debug (see section below)
 * - open serial monitor and check whats happening
 * - visit https://github.com/sinricpro/esp8266-esp32-sdk/issues and check for existing issues or open a new one
*/

// Uncomment the following line to enable serial debug output
//#define ENABLE_DEBUG

#ifdef ENABLE_DEBUG
       #define DEBUG_ESP_PORT Serial
       #define NODEBUG_WEBSOCKETS
       #define NDEBUG
#endif 

#include <Arduino.h>
#ifdef ESP8266 
       #include <ESP8266WiFi.h>
#endif 
#ifdef ESP32   
       #include <WiFi.h>
#endif

#include "SinricPro.h"
#include "SinricProLock.h"

#define WIFI_SSID         "YOUR-WIFI-SSID"    
#define WIFI_PASS         "YOUR-WIFI-PASSWORD"
#define APP_KEY           "YOUR-APP-KEY"      // Should look like "de0bxxxx-1x3x-4x3x-ax2x-5dabxxxxxxxx"
#define APP_SECRET        "YOUR-APP-SECRET"   // Should look like "5f36xxxx-x3x7-4x3x-xexe-e86724a9xxxx-4c4axxxx-3x3x-x5xe-x9x3-333d65xxxxxx"
#define LOCK_ID           "YOUR-DEVICE-ID"    // Should look like "5dc1564130xxxxxxxxxxxxxx"
#define BAUD_RATE         9600                // Change baudrate to your need

#define LED_GREEN   D6
#define LED_RED     D5

void setupWiFi() {
  Serial.printf("\r\n[Wifi]: Connecting");
  WiFi.begin(WIFI_SSID, WIFI_PASS);

  while (WiFi.status() != WL_CONNECTED) {
    Serial.printf(".");
    delay(250);
    digitalWrite(LED_BUILTIN, !digitalRead(LED_BUILTIN));
  }
  digitalWrite(LED_BUILTIN, HIGH);
  IPAddress localIP = WiFi.localIP();
  Serial.printf("connected!\r\n[WiFi]: IP-Address is %d.%d.%d.%d\r\n", localIP[0], localIP[1], localIP[2], localIP[3]);
}

bool onLockState(String deviceId, bool &lockState) {
  Serial.printf("[SmartLock]: %s\r\n", lockState?"locked":"open");
  switch (lockState) {
    case true:    digitalWrite(LED_RED, HIGH); digitalWrite(LED_GREEN, LOW);
                  break;
    case false:   digitalWrite(LED_RED, LOW); digitalWrite(LED_GREEN, HIGH);
                  break;
  }
  return true;
}

void setupSinricPro() {
  SinricProLock &myLock = SinricPro[LOCK_ID]; // This is working since verison 2.2.1 of SinricPro library. 
                                              // If you don't have 2.2.1 you have to update or use the following line instead!

//  SinricProLock &myLock = SinricPro.add<SinricProLock>(LOCK_ID); // This is for library verison below 2.2.1
  myLock.onLockState(onLockState);

  SinricPro.begin(APP_KEY, APP_SECRET);
}

void setup() {
  Serial.begin(BAUD_RATE);
  pinMode(LED_GREEN, OUTPUT); digitalWrite(LED_GREEN, HIGH);
  pinMode(LED_RED, OUTPUT);   digitalWrite(LED_RED, LOW);

  setupWiFi();
  setupSinricPro();
}

void loop() {
  SinricPro.handle();
}
```

TexAcoon, 2019-11-10:
THANK YOU! ...  I'll give it a try and let you know how I made out on my end ..
Doug

TexAcoon, 2019-11-10:
> After it is discovered, go to alexa app and configure your smart lock device. You have to explicit enable voice unlock and you have to define a pin!

Loaded the ESP8266 without issues and recognized by Alexa in discovery by Alexa.  I went to Alexa App ...  SETTINGS - DEVICES and looked for a smart lock.  I do not see where it is recognized in any of the devices the only thing on my Device List is my echo and dots ...  neither the fireplace or the garage is listed is what I have using a ESP8266.

sivar2311, 2019-11-10:
You have to login at https://sinric.pro and create a device!

TexAcoon, 2019-11-10:
YUP .. Already did ..  used Smart Lock as directed  ...  nothing shows up on my Alexa App..

sivar2311, 2019-11-10:
You need to activate SinricPro skill and link it to your SinricPro account

TexAcoon, 2019-11-10:
I did that when I did the fireplace setup

sivar2311, 2019-11-10:
Please try to add another device, like a switch or doorbell. Right after adding you should see a pop up on your smartphone from alexa app, indicating a new device has been found.

clox-coder, 2019-11-10:
@TexAcoon was curious would you mind sharing your whole code? are you using a simple relay setup?

TexAcoon, 2019-11-10:
> Please try to add another device, like a switch or doorbell. Right after adding you should see a pop up on your smartphone from alexa app, indicating a new device has been found.

Code is Great! and working as expected!  All I need to do is hook up the mechanics on my garage door and I'll be set... 

Great Job **sivar2311**!  I'm really impressed with this ...  This could be another one for **kakopappa** libraries on Sinric Pro....

TexAcoon, 2019-11-10:
clox-coder 
You should see the code above it in our comments section by sivar2311 - He really did a great job tying the Arduino code in with Sinric Pro.

clox-coder, 2019-11-10:
@TexAcoon I was just asking to make sure there was not a minor mistake. But also just in case this might help.  I would take one 8266 at a time plug into pc with power. pull up serial monitor and identify ip address assigned to device. unless you know how to identify through router. then ping the 8266 device just to verify it's on the network. if so you know your getting this far. also reset your alexa etc. it could be this simple of a problem as well.

TexAcoon, 2019-11-10:
Good Advice! .. Thanks! ..
I use checking IP addresses Angry Scanner when I have to check IP addresses ... . You do bring up a good point in order to pinpoint the or localize the ESP8266.

clox-coder, 2019-11-10:
@TexAcoon sorry your post was delayed! lol

sivar2311, 2019-11-10:
> clox-coder
> You should see the code above it in our comments section by sivar2311 - He really did a great job tying the Arduino code in with Sinric Pro.

I am the guy who is writing the library. It is very flexible... you can use any device which SinricPro offers. For detailed infos about the library, how to use it and which devices are available please see [readme in git repo](https://github.com/sinricpro/esp8266-esp32-sdk). If you encounter any issues, please use the correct git repo and check for existing issues or open a new one.

[37 more replies]