kakopappa/arduino-esp8266-alexa-wemo-switch #46: Wifi connection issue

davidallen90357, 2022-10-11:
D1 Mini will not connect to network. It has connected running examples from Arduino. Would appreciate any suggestions.

martin2149, 2022-10-11:
I used Sinric.com, Add the Sinric skill to Alexa, CVreate a Sinric account and then create devices. when your done the arduino code is created for you, all you need to do then is copy and paste the code into Arduino IDE, fill in the APP key, APP secret and device ID that sinric created for you, plug in your device, select it in the IDE and upload.
Mine worked first time. 
Good luck, I know it can be frustrating trying to get these things to work.

kakopappa, 2022-10-12:
Hi @davidallen90357 

If your D1 Mini will not connect to the WiFi network, try starting a mobile hotspot and connecting to it. Make sure you are using the latest versions of Arduino Core library.